arduino/arduino-create-agent #358: /v2/pkgs/tools/installed HTTP 500

AsaAyers, 2019-04-20:
Every time I try to upload a project to my Arduino, http://127.0.0.1:8991/v2/pkgs/tools/installed fails with an HTTP 500 and the following response:

```
{
  "name":"fault",
  "id":"UG3JaAuF",
  "message":"Create file C:\\Users\\Asa\\.arduino-create\\arduino\\avrdude\\6.3.0-arduino9: open C:\\Users\\Asa\\.arduino-create\\arduino\\avrdude\\6.3.0-arduino9: is a directory",
  "temporary":false,
  "timeout":false,
  "fault":true
}
```

When this happens the console also shows an "Uncaught (in Promise)" with the above output and Arduino Create just hangs. That "Busy" button doesn't go away until I try to cancel the upload.

[image]

Here is the data being POSTed to that URL

```
{
  "checksum": "SHA-256:f3c5cfa8d0b3b0caee81c5b35fb6acff89c342ef609bf4266734c6266a256d4f",
  "name": "avrdude",
  "packager": "arduino",
  "signature": "7628b488c7ffd21ae1ca657245751a4043c419fbab5c256a020fb53f17eb88686439f54f18e78a80b40fc2de742f79b78ed4338c959216dc8ae8279e482d2d4117eeaf34a281ce2369d1dc4356f782c0940d82610f1c892e913b637391c39e95d4d4dfe82d8dbc5350b833186a70a62c7952917481bad798a9c8b4905df91bd914fbdfd6e98ef75c8f7fb06284278da449ce05b27741d6eda156bbdb906d519ff7d7d5042379fdfc55962b3777fb9240b368552182758c297e39c72943d75d177f2dbb584b2210301250796dbe8af11f0cf06d762fe4f912294f4cdc8aff26715354cfb33010a81342fbbc438912eb424a39fc0c52a9b2bf722051a6f3b024bd",
  "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino9-i686-w64-mingw32.zip",
  "version": "6.3.0-arduino9"
}
```

I don't know Go, but I wonder if this code throws if the folder already exists on Windows. This block of code is 22 days old created by @matteosuppo . I'm on v1.1.86-5c199c1, which does have that code present.

https://github.com/arduino/arduino-create-agent/blob/5c199c1be4a9bc33a5f8d2278fa3fa51319878d3/v2/pkgs/tools.go#L80-L83

rhessus, 2019-04-21:
I have the same issue

gvarisco, 2019-05-09:
@rhessus @AsaAyers this should be fixed by now.  Could you make sure to run the latest agent version and let us know in case the problem still occurs?

closed 2019-07-10 by gvarisco