arduino/arduino-create-agent #380: Consider adding "Cofigure Origins" to the UI

joachimhs, 2019-05-20:


joachimhs, 2019-05-20:
In order for third-party developers to use the Arduino-Create-Agent for their arduino-based websites, the end-user need to edit the config.ini file and add in the domain. 

This is something most end-users won't be able to easily do, especially if the target is educational and schools. 

It would be very handy if the UI of the Create Agent enabled the end user to easily add in websites that they would like to enable to create agent for. 

[image]

umbynos, 2021-02-12:
Thank you for your suggestion, we will add it to the project backlog when we'll have some bandwidth 😁

phptuts, 2022-01-29:
@umbynos Would you be open to a pr for this?  If so would you want it to open up a webpage to configure this feature?

umbynos, 2022-02-03:
Hi @phptuts, I asked this question to our the security team and this is what they told me:
>Including a domain in the list of origin that can communicate with the arduino agent implies a high level of trust of the domain. Since we cannot ensure that in this specific case, ElectroBlocks is currently not a partner or anything alike, we should not include it in the official code.

per1234, 2022-02-08:
Closing as declined according to [the explanation provided in the other issue](https://github.com/arduino/arduino-create-agent/issues/688#issuecomment-1032525149):

>Having a custom menu item to configure just the origins is definitely not a viable solution as we are creating and exception for a specific configuration that can and should be set in the `config.ini` file.
>
>Also, adding a custom web ui has been determined not to be feasible because of [this answer](https://github.com/arduino/arduino-create-agent/issues/380#issuecomment-1029036963)

closed 2022-02-08 by per1234