arduino/arduino-create-agent #384: ArduinoCreateAgent-1.1/Arduino_Create_Bridge: error while loading shared libraries: libappindicator3.so.1: cannot open shared object file: No such file or directory

inklu, 2019-05-24:
Arduino Create Brigde plugin does not start on Debian 8 x64 after installation.
Installation process passed well. But It reports to console these messages:

home/../ArduinoCreateAgent-1.1/Arduino_Create_Bridge: error while loading shared libraries: libappindicator3.so.1: cannot open shared object file: No such file or directory
/home/.../ArduinoCreateAgent-1.1/Arduino_Create_Bridge: error while loading shared libraries: libappindicator3.so.1: cannot open shared object file: No such file or directory

facchinm, 2019-05-28:
Hi @inklu ,
one of the prerequisites of the agent is `libappindicator3`.
This library is usually installed in every distro but probably not in Debian8.
Issuing a `sudo apt-get install libappindicator3-1` should solve the issue.

inklu, 2019-05-28:
Thanks a lot!

closed 2019-05-28 by inklu