arduino/arduino-create-agent #471: Does this agent support DTR and RTS Control

kammce, 2019-10-12:
Is there anyway to control those signals, as they are used commonly with resetting boards.

curlyz, 2021-05-04:
This surprise me when there is no reset function in the agent.

BallscrewBob, 2021-05-04:
@curlyz 

There is no reset button in the regular IDE either.
However the software itself does perform a reset after upload is complete.

curlyz, 2021-05-04:
The crazy thing is that it doesn't on my Nano. The IDE do reset , but Arduino Cloud Editor doesn't, I thought maybe the team forgot to integrate it with the Agent, and nope, the Agent doesn't have the method for that. 
I think this is just the command line data passthrough the agent missing that reset parameter. Gonna investigate later.

curlyz, 2021-05-04:
I have traceback the problem to the fact that MacOS used /dev/tty and /dev/cu. When device plugged in, /tty/port and /cu/port show up. The /cu can't reset the board. And it is listed by the agent as default.