arduino/arduino-create-agent #492: Kernel panic on Mac OS Catalina

xowor, 2019-11-20:
Uploading sketches from the Arduino web Editor sometimes causes a kernel panics on my Mac. OS is Mac OS 10.15 (Catalina), and agent version is 1.1.98-fcf6b6d.

Relevant (hopefully) part of the panic report by Mac Os:
```
panic(cpu 2 caller 0xffffff8007c689ca): Kernel trap at 0xffffff8008220e87, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x00000001082e1000, CR3: 0x000000004aa5508a, CR4: 0x00000000001626e0
RAX: 0x4d39d5e090ff5d86, RBX: 0xffffff80322c1e80, RCX: 0x0000000000000000, RDX: 0x0000000000000001
RSP: 0xffffff9127af3b80, RBP: 0xffffff9127af3b80, RSI: 0xffffff8019891f20, RDI: 0xffffff80322c1e80
R8:  0x0000000000000000, R9:  0x0000000000989680, R10: 0xffffff80085f3cc0, R11: 0x0000000000000680
R12: 0x0000000000000000, R13: 0xffffff80322c1e80, R14: 0xffffff801a4abe10, R15: 0xffffff80329eb2f0
RFL: 0x0000000000010286, RIP: 0xffffff8008220e87, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x00000001082e1000, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 0

Backtrace (CPU 2), Frame : Return Address
0xffffff8007952220 : 0xffffff8007b3f98b
0xffffff8007952270 : 0xffffff8007c76c15
0xffffff80079522b0 : 0xffffff8007c6861e
0xffffff8007952300 : 0xffffff8007ae6a40
0xffffff8007952320 : 0xffffff8007b3f077
0xffffff8007952420 : 0xffffff8007b3f45b
0xffffff8007952470 : 0xffffff80082d2d89
0xffffff80079524e0 : 0xffffff8007c689ca
0xffffff8007952660 : 0xffffff8007c686c8
0xffffff80079526b0 : 0xffffff8007ae6a40
0xffffff80079526d0 : 0xffffff8008220e87
0xffffff9127af3b80 : 0xffffff7f8ad8750b
0xffffff9127af3bd0 : 0xffffff7f8995e3f3
0xffffff9127af3c10 : 0xffffff7f8995bc4d
0xffffff9127af3c40 : 0xffffff8007dd918c
0xffffff9127af3c90 : 0xffffff8007de1d66
0xffffff9127af3ce0 : 0xffffff8007dcda8a
0xffffff9127af3d30 : 0xffffff8007dbfb61
0xffffff9127af3da0 : 0xffffff8007dbf63e
0xffffff9127af3e40 : 0xffffff800805c07b
0xffffff9127af3eb0 : 0xffffff800805bcd9
0xffffff9127af3f10 : 0xffffff8008060b2d
0xffffff9127af3f40 : 0xffffff800819b999
0xffffff9127af3fa0 : 0xffffff8007ae7206
      Kernel Extensions in backtrace:
         com.apple.iokit.IOSerialFamily(11.0)[8FF5E779-7922-37DC-82D0-2A190DBF6ECF]@0xffffff7f8995b000->0xffffff7f89968fff
         com.apple.driver.usb.serial(6.0)[05F421F5-F523-374E-8249-2FF861A37D10]@0xffffff7f8ad86000->0xffffff7f8ad8bfff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[70761A70-11A9-3F83-B2F7-0FCEC749EE99]@0xffffff7f8876b000
            dependency: com.apple.iokit.IOSerialFamily(11)[8FF5E779-7922-37DC-82D0-2A190DBF6ECF]@0xffffff7f8995b000

BSD process name corresponding to current thread: bossac

Mac OS version:
19A602

Kernel version:
Darwin Kernel Version 19.0.0: Wed Sep 25 20:18:50 PDT 2019; root:xnu-6153.11.26~2/RELEASE_X86_64
Kernel UUID: 70EDD61F-86EE-3E1B-873F-98D909B78160
Kernel slide:     0x0000000007800000
Kernel text base: 0xffffff8007a00000
__HIB  text base: 0xffffff8007900000
System model name: MacBookPro11,1 (Mac-189A3D4F975D5FFC)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 367187168525
last loaded kext at 258959240000: @filesystems.msdosfs	1.10 (addr 0xffffff7f8b34d000, size 69632)
```

umbynos, 2020-11-25:
could you try with the latest version of the agent?

zmoog, 2021-01-22:
@xowor if you get a chance to do a test with the release 1.2.0 please share your results and reopen the issue.

closed 2021-01-22 by zmoog