arduino/arduino-create-agent #496: Mac Catalina 10.15.1 (19B88) [bad CPU type in executable error]

baseljd, 2019-12-02:
I tried the latest Arduino Create Agent on **Mac Catalina 10.15.1 (19B88)**. I am getting error below. 

> Start command: fork/exec /Users/Username/.arduino-create/arduino/avrdude/6.3.0-arduino14/bin/avrdude: bad CPU type in executable

There was a similar problem for the Arduino IDE, but it was fixed in this [PR 8976](https://github.com/arduino/Arduino/pull/8976).

Is there a similar fix for the Arduino Create Agent? I am guessing the Agent needs an updated Avrdude **6.3.0-arduino17** instead of **6.3.0-arduino14**.

I tried taking it from the Arduino IDE and replacing it in .arduino-create, but no luck.

gvarisco, 2019-12-02:
Hi @baseljd, any chance you can provide us the agent's version?

baseljd, 2019-12-02:
Hello @gvarisco ,

**Agent version:** 1.1.98-fcf6b6d downloaded from [here](https://downloads.arduino.cc/CreateBridgeStable/ArduinoCreateAgent-1.1-osx-installer-chrome.dmg)

atinybeardedman, 2019-12-11:
I have the same issue (also Mac Catalina) using agent 1.1.101-318de37

lukevink, 2019-12-11:
Also have the same issue on Catalina, tried manually updating as well, no joy :-(

HunterDG, 2019-12-30:
ditto - agent 1.1.101-318de37

mhansen, 2020-01-02:
Seeing this on {"Version" : "1.1.101"}

jelkinsiv, 2020-01-03:
Same here - agent 1.1.101-318de37

QiuFeng54321, 2020-01-04:
Ditto - 1.1

QiuFeng54321, 2020-01-04:
I've compared [PR8976](https://github.com/arduino/Arduino/pull/8976) described by @baseljd , The version changes of package indexes in https://github.com/arduino/Arduino/pull/8976/files#diff-fe2506017d5743b1f425f8c1a70c2dfe are not applied to https://downloads.arduino.cc/packages/package_index.json and https://github.com/arduino/arduino-create-agent/blob/devel/v2/pkgs/testdata/package_index.json (Not sure if this matters because the file path contains `testdata`.  
The version of avr, avr-gcc are outdated(**1.6.2** instead of the **1.8.1**(or newer))   
avrdude in https://downloads.arduino.cc/packages/package_index.json : 
```json
{
  "packager": "arduino",
  "name": "avrdude",
  "version": "6.0.1-arduino3"
}
```
avrdude in https://github.com/facchinm/Arduino/blob/3af92cdfc8a90874461f021bdc397c29a78dd23f/hardware/package_index_bundled.json : 
```json
{
   "packager": "arduino",
   "name": "avrdude",
   "version": "6.3.0-arduino17"
},
```

mpwteach, 2020-01-08:
Same here - please fix this bug - agent 1.1.101-318de37, latest macOS Catalina...

facchinm, 2020-01-09:
@QiuFeng54321 https://downloads.arduino.cc/packages/package_index.json  contains the right packages. Tool dependencies for avr code 1.8.1 : 
```
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino5"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino17"
            },
            {
              "packager": "arduino",
              "name": "arduinoOTA",
              "version": "1.3.0"
            }
          ]
        },
```

theChris1310, 2020-01-09:
I got the same issue. Uninstalles everything, deleted all related files in `.Library/` and rebooted. New install of Arduino Create Agent. And as soon as I try to upload an file, the Version 6.3.0-arduino14 is installed in `.arduino-create/arduino/avrdude/`

Is there a possibility to change the download/update URL to the package in the Arduino Create Agent?

@facchinm, how should your reply help me?

`Catalina (10.15.2), Create Agent shows {"Version" : "1.1.101"} in debug console.`

QiuFeng54321, 2020-01-11:
@facchinm It seems that the package_index.json is not for arduino create agent.
There is an avrdude 6.3.0-arduino17 version for OSX and nobody has that.
In addition It seems that avr-gcc is outdated now. I cannot find pattern /avr-gcc.*arduino17/ in the file.
I am confused about package_index.json. I will read some documents about it.

gvarisco, 2020-01-13:
Hi everyone! Any chance you can give it another try at Create + Create Agent on macOS Catalina and let us know if the problem still persists?

mpwteach, 2020-01-13:
I just tried, but it downloads the same version from the Arduino website and it is not fixed.  How do I download the new version?

rsora, 2020-01-14:
Hi @mpwteach, thanks for checking it again and commenting here. 
The issue was caused by a misalignment of the core versions in the Create Backend services, so the problem was not an Agent problem. 
We refreshed now the core installations on the backend, Can you try again and, in case of errors, paste here the output message printed by the editor?

Thanks a lot!

baseljd, 2020-01-14:
Hi @rsora , I just tested it. 

Deleted the .arduino-create agent folder in the Home directory that already had the old avrdude14 and then reuploaded and I got avrdude17.

Uploading works fine now! 🚀 

Thanks!

closed 2020-01-14 by baseljd

mpwteach, 2020-01-14:
Thanks, it now works!  Not sure it this is related, but the Chromebook version of Arduino Create is completely broken as of yesterday...  There are numerous posts on this issue here:
https://forum.arduino.cc/index.php?board=101.0

rsora, 2020-01-14:
@mpwteach Thanks for the reporting, I have forwarded your warning to the chrome app team, They will reply directly in the forum posts!

mpwteach, 2020-01-14:
Sorry, just checked on a Chromebook, looks like they fixed the issue last night, so all is good!  Thanks!