arduino/arduino-create-agent #515: Launching / Installing fail https handler fail

lostsh, 2020-01-19:
# Instalation of Arduino Create Agent on Debian 10

Installation of the plugin does not seem to work, however, it can be noted that the process proceeds without apparent problems, and that no error messages are returned, however the plugin does not work.

per1234, 2020-01-19:
Duplicate of https://github.com/arduino/Arduino/issues/9639

umbynos, 2021-02-12:
Hi, could you please tell us what steps did you followed? It would be really helpful...
What version are you using? What specific problem did you stumbled upon? 
We use installbuilder to generate the installer.
The installer generates a logfile inside `/tmp/installbuilder_installer_XXXX.log`, could you please include it in the issue?

closed 2021-03-08 by github-actions[bot]