arduino/arduino-create-agent #552: "go get" command fails on Raspberry Pi 4

electrontinkerer, 2020-04-16:
I am unable to download and install packages and their dependencies due to problems in the source files. The errors reported by go are:

# github.com/arduino/arduino-create-agent/gen/http/indexes/server
go/src/github.com/arduino/arduino-create-agent/gen/http/indexes/server/encode_decode.go:35:37: not enough arguments in call to "goa.design/goa/http".ErrorEncoder
	have (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder)
	want (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder, func(error) "goa.design/goa/http".Statuser)
go/src/github.com/arduino/arduino-create-agent/gen/http/indexes/server/encode_decode.go:95:37: not enough arguments in call to "goa.design/goa/http".ErrorEncoder
	have (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder)
	want (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder, func(error) "goa.design/goa/http".Statuser)
go/src/github.com/arduino/arduino-create-agent/gen/http/indexes/server/encode_decode.go:155:37: not enough arguments in call to "goa.design/goa/http".ErrorEncoder
	have (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder)
	want (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder, func(error) "goa.design/goa/http".Statuser)
# github.com/arduino/arduino-create-agent/gen/http/tools/server
go/src/github.com/arduino/arduino-create-agent/gen/http/tools/server/server.go:109:40: not enough arguments in call to "goa.design/goa/http".ErrorEncoder
	have (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder)
	want (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder, func(error) "goa.design/goa/http".Statuser)
go/src/github.com/arduino/arduino-create-agent/gen/http/tools/server/server.go:153:40: not enough arguments in call to "goa.design/goa/http".ErrorEncoder
	have (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder)
	want (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder, func(error) "goa.design/goa/http".Statuser)
go/src/github.com/arduino/arduino-create-agent/gen/http/tools/server/server.go:198:40: not enough arguments in call to "goa.design/goa/http".ErrorEncoder
	have (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder)
	want (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder, func(error) "goa.design/goa/http".Statuser)
go/src/github.com/arduino/arduino-create-agent/gen/http/tools/server/server.go:250:40: not enough arguments in call to "goa.design/goa/http".ErrorEncoder
	have (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder)
	want (func(context.Context, "net/http".ResponseWriter) "goa.design/goa/http".Encoder, func(error) "goa.design/goa/http".Statuser)

mundostr, 2020-06-25:
Hello:
I am experiencing the same issue in RPi3b, any updates on this will be appreciated.

sorydi3, 2020-12-27:
having the same issues as well.

dhruvsheth-ai, 2021-01-19:
I am getting the same error, any decoding options would be helpful @facchinm , @rsora 
Thanks

zmoog, 2021-01-22:
Hello, we successfully tested the building of the Agent on a Raspberry Pi 3 Model B. using a Raspbian GNU/Linux 10 (buster).

Here’s the commands used to build it:

```shell
$ sudo apt install golang

$ git clone https://github.com/arduino/arduino-create-agent.git

$ cd arduino-create-agent/

$ go build -v -i -tags cli

$ ./arduino-create-agent
INFO[0000] Version:x.x.x-dev
INFO[0000] Hostname: raspberrypi
INFO[0000] Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect.
INFO[0000] You specified a serial port regular expression filter: usb|acm|com
INFO[0000] Your serial ports:
INFO[0000] 	There are no serial ports to list.
[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:	export GIN_MODE=release
 - using code:	gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET    /                         --> main.homeHandler (2 handlers)
[GIN-debug] GET    /certificate.crt          --> main.certHandler (2 handlers)
[GIN-debug] DELETE /certificate.crt          --> main.deleteCertHandler (2 handlers)
[GIN-debug] POST   /upload                   --> main.uploadHandler (2 handlers)
[GIN-debug] GET    /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] POST   /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] WS     /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] WSS    /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] GET    /info                     --> main.infoHandler (2 handlers)
[GIN-debug] POST   /killbrowser              --> main.killBrowserHandler (2 handlers)
[GIN-debug] POST   /pause                    --> main.pauseHandler (2 handlers)
[GIN-debug] POST   /update                   --> main.updateHandler (2 handlers)
[GIN-debug] GET    /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] POST   /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] PUT    /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] PATCH  /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] HEAD   /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] OPTIONS /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] DELETE /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] CONNECT /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] TRACE  /v2/*path                 --> github.com/gin-gonic/gin.WrapH.func1 (2 handlers)
[GIN-debug] Listening and serving HTTP on 127.0.0.1:8991
ERRO[0000] Could not find HTTPS certificate. Using plain HTTP only.
```

Can you share your equivalent shell commands?

dhruvsheth-ai, 2021-01-22:
Hi, I tried using the commands you have given but the below command - 
`./arduino-create-agent`gives an error:
bash: ./arduino-create-agent: No such file or directory

Do I have to install the .tar.gz file for the plugin, unzip it and then run the command?

P.S: I have used the above command from arduino-create-agent directory on terminal
[image]

umbynos, 2021-01-22:
Hi, please try building with `sudo`: `sudo go build -v -i -tags cli`
to check if the compiled binary is present in the folder you can use `ls`.

By the way I just tried it on my raspberry Pi 4 with Raspbian 10 buster and it works :partying_face:

umbynos, 2021-01-22:
You can run it with the `&` operator and `nohup`: `nohup ./arduino-create-agent &`
- `&`starts the program in background
- `nohup` cashes the hangup signal and when the parent process is killed (the shell) it changes the parent to 1 (the init process)

umbynos, 2021-01-22:
Closing because the problem apparently is solved. Feel free to re-open it

closed 2021-01-22 by umbynos

dhruvsheth-ai, 2021-01-22:
Awesome, Thanks!

umbynos, 2021-01-25:
FYI I've created a [wiki page](https://github.com/arduino/arduino-create-agent/wiki/How-to-compile-on-Raspberry-Pi)