arduino/arduino-create-agent #554: Better Linux re-packaing compatibility

tinywrkb, 2020-05-14:
There are a few issues that prevent re-packaging and distributing Arduino Create Agent through repositories instead of the current binary installer.

There are blocking issues that prevent repacking:

* No tarball release.  
The binary installer, even with the `--mode unattended` still tries to run Arduino_Create_Bridge after the installation and it does not respect `--prefix`.  
* Executable and config in the same folder limitation.  
The app should look for the config at `$XDG_CONFIG_HOME/arduino-create/` with `$HOME/.config/arduino-create/` as a fallback if `$XDG_CONFIG_HOME` is unset or empty. See [XDG Base Directory Specification](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html)

Would be nice to have:

* Instead of `$HOME/.arduino-create/` use `$XDG_DATA_HOME/arduino-create/` with `$HOME/.local/share/arduino-create/` as a fallback if `$XDG_DATA_HOME` is unset or empty.
* In case that `config.ini` does not exist in the config folder, create a default config file.
* A single app for both Chrome and Firefox. The current Linux installers are actually the same for both Chrome and Firefox, I don't know if this is intentional.

For my own needs, and as a POC, I packaged the Create Agent as a Flatpak [here](https://github.com/tinywrkb/flatpaks/blob/master/cc.arduino.arduino-create-agent/cc.arduino.arduino-create-agent.yaml). Notice the ugly workaround in [the startup script](https://github.com/tinywrkb/flatpaks/blob/master/cc.arduino.arduino-create-agent/resources/arduino-create-agent), copying the executable to the config folder.

umbynos, 2020-11-06:
Hello! Thanks for the interest in the project!
At the moment we are in the middle of a lift & shift process regarding the CI/CD pipeline to generate the installers. We will for sure consider the POC you made in the near future.

tinywrkb, 2020-11-07:
@umbynos, thank you for your reply. I hope that you won't only make it easy for package maintainers to re-package the app but also consider distribution via Flathub as a Flatpak app.

tinywrkb, 2021-04-08:
Any news about this?

Can we at least get a binary tarball release? The installer needs access to `/proc/self/exe` and due to the security implications of having such access, it's not available during the Flatpak installation/apply_extra step as Flatpak is running as the root user.  
I'm seeing a failed readlink(/proc/self/exe) and the installer.  
I can go around this issue by installing to the user Flatpak installation and not system-wide but that's not a proper solution.

Also, please consider officially distribute this app via Flathub.

Note that I'm not that sure about certificate handling. I guess maybe bind-mount `~/.pki/nss` from the host, package `certutil`, and add support in the app for adding certificate via `certutil`.  
I should also add that IIRC correctly Firefox already support the XDG dirs specification for the certs DB while Chrome/Chromium doesn't.