arduino/arduino-create-agent #566: Crash during upload with serial monitor open

umbynos, 2020-11-06:
## Bug Report

### Describe the bug

crash
### To Reproduce

Upload with serial monitor open sometimes causes a crash
### Expected behavior

no crash
## Environment (please complete the following information):
 - OS [e.g. Windows]: Ubuntu
 - OS version: 20.04.1 LTS
 - Browser [e.g. chrome, safari]: Chrome
 - Browser Version: 86.0.4240.111
 - Agent Version [e.g. 1.0.0]: 1.1.251

## Additional context

```
INFO[1211] {"uploadStatus": "Killed"}                   
panic: send on closed channel
goroutine 4051 [running]:
main.(*BufferflowTimed).Close(0xc0001b4ac0)
	/home/jenkins/workspace/Create_agent_master/src/github.com/arduino/arduino-create-agent/bufferflow_timed.go:110 +0x54
main.spCloseReal(0xc000290ba0)
	/home/jenkins/workspace/Create_agent_master/src/github.com/arduino/arduino-create-agent/serialport.go:364 +0x34
main.spHandlerClose(...)
	/home/jenkins/workspace/Create_agent_master/src/github.com/arduino/arduino-create-agent/serialport.go:360
main.spClose(0xc000146166, 0xc)
	/home/jenkins/workspace/Create_agent_master/src/github.com/arduino/arduino-create-agent/serial.go:241 +0x117
created by main.checkCmd
	/home/jenkins/workspace/Create_agent_master/src/github.com/arduino/arduino-create-agent/hub.go:146 +0x7f2
```

Webbeh, 2020-12-01:
I do have the same all the time.
Windows 10 Pro, and the plugin doesn't crash when serial monitor is off.

Furthermore, after crashing and re-opening the agent, compilations/uploads take a really long time unless you also refresh the arduino create webpage.

Rhragnarok, 2021-01-15:
I need this fixed asap.  Is there any info on when it might be fixed?  Kinda makes it hard to work.

umbynos, 2021-01-18:
:wave: We just released [1.2.0](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.0) Could you please try that version?

Webbeh, 2021-01-18:
I had the same issue on 1.2.0-rc3 45a2de6. Is there any change between the release and the RC3 ?

BallscrewBob, 2021-01-18:
It crashed out here too mid install.

Webbeh, 2021-01-18:
The only behaviour that I am absolutely sure of, for both 1.2.0 and the versions before 75865e6, the crashes only happen if we have the serial monitor tab open, or if we switched from/to the monitor tab a few seconds before sending the program. It never crashed on sessions where I didn't use the serial monitor.

umbynos, 2021-01-18:
Thanks for the feedback, I'll try to solve that issue as soon as possible.

Could you please provide some context on the environment used?

    OS [e.g. Windows]:
    OS version:
    Browser [e.g. chrome, safari]:
    Browser Version:
    Agent Version [e.g. 1.0.0]:

umbynos, 2021-01-18:
We implemented a crashreport functionality to better address these problems. Please enable it and include the reports in the PR to help us :smile: 
https://github.com/arduino/arduino-create-agent/wiki/How-to-use-crashreport-functionality

umbynos, 2021-01-29:
Hi guys, A user reported a similar bug here #609 
It should be fixed. Could you please try [1.2.1-rc1](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.1-rc1)? Let me know if it's working for you

Webbeh, 2021-01-30:
Seems to have resolved the issue for me so far, after ~25 uploads.

umbynos, 2021-02-01:
Perfect! :space_invader: Closing as the issue seems to be solved. Feel free to reopen

closed 2021-02-01 by umbynos