arduino/arduino-create-agent #568: Error reported yet appears to complete.

BallscrewBob, 2020-11-07:
## Bug Report

### Describe the bug
Install error

### To Reproduce
Install create plugin from the FLOW procedure

### Expected behavior
Complete without error

## Environment (please complete the following information):
 - OS win 10 pro
 - OS version
 - Browser Latest Chrome
 - Browser Version: Version 86.0.4240.183 (Official Build) (64-bit)
 - Agent Version [e.g. 1.0.0]: 1.1.251-318de37

## Additional context
Web page showed a successful install but the process had not completed
[image]

umbynos, 2020-11-18:
It's failing on every install?

BallscrewBob, 2020-11-18:
Not any more so I think this can be closed as I used the latest version which I think you may have pushed some commits up to ?

umbynos, 2020-11-18:
Thanks for trying the new version! 1.2.0-dev actually is a prerelease beta version so it could contain bugs and it's not ready for production yet. But it's good for you to beta-test it. We have replaced the CI pipeline to generate the installers (if you are curious [here](https://github.com/arduino/arduino-create-agent/blob/devel/.github/workflows/release.yml)'s the workflow). In the next weeks, we are going to do a release in production using the new workflow. Then we are going to fix bugs and issues and test new features introduced on the branch devel and have not yet been part of a release.

closed 2020-11-18 by umbynos