arduino/arduino-create-agent #571: Create Agent Beta testing

jojo1212, 2020-11-25:
## Bug Report

### Describe the bug

When running the setup to add device to IoT cloud during the 3 rd step gives me an error as the one below
"Uncaught SyntaxError: Unexpected token p in JSON at position 0 "
then keeps running the 3rd step.

### To Reproduce
- MKR 1010 attached to the computer with USB cable (no other attachments)
- Add device to the dev version of IoT cloud wait till the 3rd step

### Expected behavior
Pass the 3rd step and adding the device to IoT cloud ready to connect to thing.

## Environment (please complete the following information):
 - OS [e.g. Windows]: Windows
 - OS version: 10 Home 2004
 - Browser [e.g. chrome, safari]: Chrome
 - Browser Version: 87.0.4280.66 (Official Build) (64-bit)
 - Agent Version [e.g. 1.0.0]: ArduinoCreateAgent-1.2.0-dev-windows-installer-chrome

## Additional context

Made sure to delete the production version of the Create agent.

umbynos, 2020-11-25:
It's working on the prod version of create?

> Made sure to delete the production version of the Create agent.

What do you mean for this?

jojo1212, 2020-11-25:
I meant I had only the beta version installed in the system.

umbynos, 2021-01-18:
Could you please try with the latest version? [1.2.0](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.0)

umbynos, 2021-01-22:
Any feedback?

zmoog, 2021-01-26:
Hey @jojo1212, let us know how it's going with [1.2.0](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.0) or [1.2.1-dev](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.1-dev) and feel free to reopen the issue if you have something new.

closed 2021-01-26 by zmoog