arduino/arduino-create-agent #572: Re-opening a prev issue.

BallscrewBob, 2020-11-25:
## crash during install of PROD version

### Describe the bug
crash during install of PROD version

### To Reproduce
Install latest version from flow link

### Expected behavior
No errors

## Environment (please complete the following information):
 - Win 10 x 64 pro 
 - Build 19041
 - Browser  chrome, 
 - Browser Version: Version 87.0.4280.66 (Official Build) (64-bit)
 - Agent Version [e.g. 1.0.0]: 1.1.25-318 de 37

## Additional context
[image]

No other information available.

umbynos, 2021-01-22:
Please could you try with the latest version? [1.2.0](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.0)
Also please try to uninstall the previously installed version before installing the new one

I tried to replicate your problem, but the installation succeeds.

BallscrewBob, 2021-01-22:
Avast still warns of the alexa virus.
I know this is an FP event but it removes and prevents it from running.

Win 10 x64 pro.

zmoog, 2021-01-22:
@BallscrewBob is the warning on the installer or the Agent itself?

BallscrewBob, 2021-01-22:
[image]
 The installer gives an alexa virus warning but if I try to run the PROD installer I get the one above for a generic virus.

BallscrewBob, 2021-01-22:
You have to make an EXCEPTION for it.
After making an exception I get this.
[image]

BallscrewBob, 2021-01-22:
After the above error I also get the following.
Pretty sure it is linked to the AVAST issue

[image]

BallscrewBob, 2021-01-22:
CONFIRMED it was avast by disabling avast and running the installer as administrator.

BallscrewBob, 2021-01-22:
HEADS UP was using 1.2 for the last few posts above.

zmoog, 2021-01-22:
After you disabled Avast, were you able to successfully run the installer and the Agent?

BallscrewBob, 2021-01-22:
I was able to INSTALL the agent but the agent itself kept closing (seems random or during an upload) leaving me with unable to upload and the message telling me to install the agent comes up in the editor.

BallscrewBob, 2021-01-22:
Version listed is 1.2.0-5C796D2 btw.

umbynos, 2021-01-25:
Could you please enable the crashreport? https://github.com/arduino/arduino-create-agent/wiki/How-to-use-crashreport-functionality

BallscrewBob, 2021-02-05:
Can be closed as was traced down to AVAST AV issues.

closed 2021-02-05 by BallscrewBob