arduino/arduino-create-agent #609: Serial connection closes after receiving `interrupted system call`

petrica, 2021-01-28:
## Bug Report

### Describe the bug
After opening the serial monitor and reading data from an Arduino, if the amount of data is consistent and protracted in time, an interrupted system call error closes the serial connection.

### To Reproduce
Write an Arduino sketch that sends data continuously to the serial monitor.

### Expected behavior
Connection is kept opened and data is received.

## Environment (please complete the following information):
 - OS: macOs
 - OS Version: Catalina 10.15.7
 - Browser: Chrome
 - Browser Version: 87.0.4280.141
 - Agent Version [e.g. 1.0.0]: 1.2.0-5c796d2

## Additional context
This bug seems to have been present in the Arduino CLI as well and got resolved by upgrading the bugst/go-serial library to latest version.

Issue fixed here: https://github.com/arduino/arduino-cli/pull/507

zmoog, 2021-01-29:
Hey @petrica thank you for the troubleshooting, kudos!

We're opening a PR to build a test version of the Agent so you can test it..

petrica, 2021-01-29:
Sure, I have already downloaded the source code and tested it replacing the `go.bug.st/serial.v1` import with `go.bug.st/serial`. Interestingly enough, although `go.bug.st/serial.v1` refers to v1 of the library, it does not reference the code under v1.0.0 where the fix is.

zmoog, 2021-01-29:
We have a 1.2.1 release coming so if your test is successful we can ship it to all users.

Thank you! 🙇

umbynos, 2021-01-29:
https://github.com/arduino/arduino-create-agent/actions/runs/521166778
Here's the artifacts, download the binaries and please tell us if this solves the problem :bug: :zap:

petrica, 2021-01-29:
Running this successfully for the past 10 minutes. Usually it stopped in less than a minute.

Tested arduino-create-agent-macos-10.15-amd64.zip artifact on MacOS Catalina 10.15.7.

Thank you for the quick reaction to this issue!

umbynos, 2021-01-29:
Thank you for taking the time to report a so detailed issue!! :love_letter:
I will merge the PR right now! This fix will be part of the 1.2.1 incoming release.

closed 2021-01-29 by umbynos