arduino/arduino-create-agent #611: List is empty

dimensi, 2021-02-02:
## Bug Report

### Describe the bug
Hello, the agent can't see the ports.

### To Reproduce
Start the agent. Open debug console. Print "list". 

### Expected behavior
List with any ports

## Environment (please complete the following information):
 - OS [e.g. Windows]: Mac OS 11.1
 - Browser [e.g. chrome, safari]: Chrome 
 - Browser Version: 88
 - Agent Version [e.g. 1.0.0]: 1.2.0

## Additional context

[image]

Arduino Ide see the ports 
[image]

umbynos, 2021-02-02:
Hello! Are you using the M1 mac?

dimensi, 2021-02-02:
Yep. Mac mini m1

umbynos, 2021-02-03:
Unfortunately, the agent does not work out of the box with the new Apple architecture as stated in the [wiki](https://github.com/arduino/arduino-create-agent/wiki#apple-silicon-macs-support). We will add support for it hopefully in the near future

closed 2021-02-03 by umbynos