arduino/arduino-create-agent #612: Error: ArduinoIoTCloudTCP::handle_ConnectMqttBroker could not connect to mqtts-sa.iot.arduino.cc:8883

jojo1212, 2021-02-04:
## Bug Report

### Describe the bug
This error usually shows up randomly and oce it starts to show up it never goes away.

### To Reproduce
Randomly comes up, even if you follow the procedure properly.

### Expected behavior
No connection to Arduino IoT cloud.

## Environment (please complete the following information):
 - OS [e.g. Windows]: Windows
 - OS version: 10.0.19041 Build 19041
 - Browser [e.g. chrome, safari]: Chrome
 - Browser Version: 88.0.4324.104 (Official Build) (64-bit)
 - Agent Version [e.g. 1.0.0]: 1.2.1-rc1
 - CPU architecture [e.g. amd64]: AMD

## Additional context
I have got this error multiple times and ususally I solved this error my reinstalling the agent again and then adding the device as a new board.

umbynos, 2021-02-04:
Could you please provide some logs/crashreports? Were do you see that error?

jojo1212, 2021-02-04:
This error shows up on the serial Monitor after uploading the sketch. I was using an MKR 1010 at the moment of testing. I did not copy the debug console window at that time when I got the error. But one of the customer with the same issue shared his Create agent debug console.

```
> {"Version" : "1.2.0"}
> {"Commands" : ["list", "open [portName] [baud] [bufferAlgorithm (optional)]", "send [portName] [cmd]", "sendnobuf [portName] [cmd]", "close [portName]", "bufferalgorithms", "baudrates", "restart", "exit", "program [portName] [board:name] [$path/to/filename/without/extension]", "programfromurl [portName] [board:name] [urlToHexFile]"]}
> {"Hostname" : "mycomputername"}
> {"OS" : "darwin"}
> list
> { "Ports": [], "Network": true }
> { "Ports": [ { "Name": "/dev/cu.usbmodem2201", "SerialNumber": "211544AD50534D53302E3120FF192615", "DeviceClass": "", "IsOpen": false, "IsPrimary": false, "Baud": 0, "BufferAlgorithm": "", "Ver": "1.2.0", "NetworkPort": false, "VendorID": "0x2341", "ProductID": "0x8054" } ], "Network": false }
```

zmoog, 2021-02-04:
> Error: ArduinoIoTCloudTCP::handle_ConnectMqttBroker could not connect to mqtts-sa.iot.arduino.cc:8883

@jojo1212 this looks like an error from the sketch running on the board, right?

jojo1212, 2021-02-04:
@zmoog the sketch I used a skecth which was working before from opla kit. Later when I uninsatlled the agent, deleted the device and reinstalled the agent again everything worked fine. I didn't make any changes to the sketch other than associating a board.

umbynos, 2021-02-05:
Closing because this is not an issue with the agent. The error `Error: ArduinoIoTCloudTCP::handle_ConnectMqttBroker could not connect to mqtts-sa.iot.arduino.cc:8883` seems to come from the ArduinoIoTCloud lib [here](https://github.com/arduino-libraries/ArduinoIoTCloud/blob/60cdc4d214dbe4534c2472e1f1319c2a9fc486f8/src/ArduinoIoTCloudTCP.cpp#L280). The task of the agent is simply to upload code to the board and read the serial from it. This error seems to come from the serial, as far as I know the agent seems to be working fine

closed 2021-02-05 by umbynos

Innov8-IT-Biz, 2021-02-06:
I am getting the same error in the serial monitor. The nano 33 iot seems to disconnect from wifi every now and then and eventually just stalls. I tried including the state machine program by Klaus_K https://forum.arduino.cc/index.php?topic=664834.15 but still having issues. I need this device to be connected remotely 24x7. Anyhelp would be appreciated.