arduino/arduino-create-agent #632: Cannot run arduino create agent

TheBluePhoenix10, 2021-04-12:
If the arduino create agent stops by chance, the icon disappears from my system tray and I cannot find any application to restart the application

umbynos, 2021-04-13:
Hi, could you please provide the context specified in the [issue template](https://github.com/arduino/arduino-create-agent/blob/main/.github/ISSUE_TEMPLATE/bug_report.md)?
Could you please provide us with a [crashreport](https://github.com/arduino/arduino-create-agent/wiki/How-to-use-crashreport-functionality)

g3holliday, 2021-05-19:
I have a RPI 4 and I executed the `./arduino-create-agent` code after compiling, but all I got was a `bash: ./arduino-create-agent: Is a directory` .

I apologize if this is a "noob" question. Thank you very much,

g3holliday

g3holliday, 2021-05-19:
The `ls -lah` returned this:

total 348K
drwxr-xr-x 17 pi pi 4.0K May 19 00:49 .
drwxr-xr-x 30 pi pi 4.0K May 19 00:55 ..
-rw-r--r--  1 pi pi 1.5K May 19 00:49 bufferflow_default.go
-rw-r--r--  1 pi pi 1.8K May 19 00:49 bufferflow.go
-rw-r--r--  1 pi pi 2.2K May 19 00:49 bufferflow_timed.go
-rw-r--r--  1 pi pi 2.1K May 19 00:49 bufferflow_timedraw.go
-rw-r--r--  1 pi pi  30K May 19 00:49 certificates.go
-rw-r--r--  1 pi pi  978 May 19 00:49 config.ini
-rw-r--r--  1 pi pi 5.7K May 19 00:49 conn.go
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 design
-rw-r--r--  1 pi pi 4.2K May 19 00:49 discovery.go
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 docs
-rw-r--r--  1 pi pi  855 May 19 00:49 .drone.sec
-rw-r--r--  1 pi pi  149 May 19 00:49 .drone.sec.example
-rw-r--r--  1 pi pi 2.4K May 19 00:49 .drone.yml
drwxr-xr-x  6 pi pi 4.0K May 19 00:49 gen
drwxr-xr-x  8 pi pi 4.0K May 19 00:49 .git
drwxr-xr-x  4 pi pi 4.0K May 19 00:49 .github
-rw-r--r--  1 pi pi  166 May 19 00:49 .gitignore
-rw-r--r--  1 pi pi 2.5K May 19 00:49 go.mod
-rw-r--r--  1 pi pi  52K May 19 00:49 go.sum
-rw-r--r--  1 pi pi 6.6K May 19 00:49 home.html
-rwxr-xr-x  1 pi pi 7.0K May 19 00:49 hub.go
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 icon
drwxr-xr-x  5 pi pi 4.0K May 19 00:49 images
-rw-r--r--  1 pi pi  744 May 19 00:49 info.go
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 killbrowser
-rw-r--r--  1 pi pi  957 May 19 00:49 killbrowser.go
-rw-r--r--  1 pi pi  18K May 19 00:49 LICENSE.md
-rwxr-xr-x  1 pi pi  20K May 19 00:49 main.go
-rw-r--r--  1 pi pi  878 May 19 00:49 manifest.xml
-rw-r--r--  1 pi pi  13K May 19 00:49 poetry.lock
-rw-r--r--  1 pi pi  349 May 19 00:49 .project
-rw-r--r--  1 pi pi  636 May 19 00:49 pyproject.toml
-rw-r--r--  1 pi pi 6.2K May 19 00:49 README.md
-rw-r--r--  1 pi pi  337 May 19 00:49 redirect_stderr_unix.go
-rw-r--r--  1 pi pi  870 May 19 00:49 redirect_stderr_win.go
-rwxr-xr-x  1 pi pi 7.2K May 19 00:49 serial.go
-rwxr-xr-x  1 pi pi 2.1K May 19 00:49 seriallist.go
-rwxr-xr-x  1 pi pi  11K May 19 00:49 serialport.go
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 systray
-rw-r--r--  1 pi pi 2.4K May 19 00:49 Taskfile.yml
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 test
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 tools
-rw-r--r--  1 pi pi 2.0K May 19 00:49 update.go
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 updater
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 upload
drwxr-xr-x  2 pi pi 4.0K May 19 00:49 utilities
drwxr-xr-x  3 pi pi 4.0K May 19 00:49 v2

Thanks!

umbynos, 2021-05-20:
Well it seems the compiled binary is not present in the directory.. :sweat_smile: Please make sure to follow the guide I linked.
https://github.com/arduino/arduino-create-agent/wiki/How-to-compile-on-Raspberry-Pi

closed 2021-06-21 by github-actions[bot]