arduino/arduino-create-agent #634: Arduino Create Agent won't install

delucadana, 2021-04-27:
I have Arduino IDE 1.8.6 and had Arduino Create agent 1.1 running on my iMac.  It worked great with my Arduino Uno.  Just bought a UNO WIFI R2 board and connected it (USB) to the same iMac running High Sierra 10.13.6.  The board lights up and it's cycling through a rainbow of very bright LED colors.   The IDE sees the board.  Under the Tools menu it identifies it as "Board:  Arduino Uno Wifi Rev 2" and Port: "/dev/cu,usbmodemFD132 (Arduino Uno WiFi Rev 2)".  I went to Arduino IOT cloud using Safari 13.1.2 at URL https://create.arduino.cc/iot/things and clicked on ASSOCIATE DEVICE, then clicked Set Up an Arduino Device.  It tells me "Arduino Create Agent not found" yet the Arduino Create Agent icon is sitting up in the menu bar indicating it's running.  I checked with the Activity Monitor app and confirmed under the CPU tab that Arduino Create Agent was active.  However the Arduino Create Agent running was version 1.1.  The IOT cloud web page says I need to download and install the Arduino Create Agent plugin in order to upload sketches to the board.  So I downloaded it, and noticed it was vers 1.2.2. Tried to install the new agent but got nothing except the warning that it was an application downloaded from internet.  Clicked OPEN and nothing happens.  The old Arduino Create agent was still running though.    And IOT Cloud web page still says Arduino Create Agent not found.  Did a restart after the new install.  No different.  So figured I needed first to uninstall the old vers 1.1 Arduino Create Agent, which I did.  Then ran the new agent installer again.  Still no install.  There's no feedback and the old agent vers 1.1  is no longer running, so I doubt even the original UNO will work anymore.  The old agent had an icon in my launcher window that's gone now after the uninstall.  No new icon has appeared for the new agent vers 1.2.2.   Any ideas on what else I should try?  Thanks.  Sorry for the length but I'm trying to provide enough info so you don't have to ask for it.

delucadana, 2021-04-27:
I think I'll try it with Chrome even though I don't like that browser because it dropped an annoying Software Updater daemon on my computer which just occupies CPU time and gets hung up every once in a while slowing things down but that's all beside the point. :-)

delucadana, 2021-04-27:
Well I couldn't get myself to download evil Chrome again after having removed it long ago.  So I tried Opera.  It does not tell me I need to download the Arduino Create Plugin. Instead it just gives me a blank stare, which is to say it puts up a blank page and sits there as if to say, huh?

delucadana, 2021-04-27:
I tried FireFox next.  This browser acts the same way Safari did.  It tells me Create Agent not found.  I down loaded the Firefox plugin and nothing happens.  It still says Create Agent not found.

BallscrewBob, 2021-04-27:
There is a [dedicated forum section](https://forum.arduino.cc/c/software/editor/134) also for these issues and you might find an answer there

umbynos, 2021-04-27:
Hi, could you please try to download and run [this](https://github.com/arduino/arduino-create-agent/releases/download/1.2.2/ArduinoCreateAgent-1.2.2-osx-amd64-installer-safari.dmg)?

delucadana, 2021-04-27:
umbynos, Hi thanks.  Yes I downloaded your link and see that the file as it appears in my downloads folder is identical to one that I downloaded six times.  I tried it again with same result.  I saw a post today for 2019 from someone who had the same problem.  It that post, he said he had recently run CleanMyMac software and eliminated 6GB of"junk".  Well it so happens that two days ago I ran CleanMyMac too.  So it maybe that the clean-up program deleted some file critical to the plugin installation.

delucadana, 2021-04-27:
> There is a [dedicated forum section](https://forum.arduino.cc/c/software/editor/134) also for these issues and you might find an answer there

Bob, yes in fact I posted this similar question there.  https://forum.arduino.cc/t/cant-install-arduino-create-agent-on-macos/854536

delucadana, 2021-04-27:
Next move for me will be to connect the Arduino Wifi Rev 2 to my PC and try the whole operation there.

umbynos, 2021-04-28:
As far as I know the installer is self-contained and should not need anything special from the host system.. We use [installbuilder](https://installbuilder.com/) to generate the installers. I tried a quick google search about compatibility problems between that program and CleanMyMac and could not find anything useful. I know that the installer tries to write a log in `/tmp` with the name `bitrock_installer.log ` as stated in the docs:
>All InstallBuilder installers create an installation log in the system temporary directory.

>The exact location of the log file is stored in the ${installer_installation_log} variable. On Linux, Mac OS X and other Unix systems, this typically means the /tmp directory. On Windows, the log will be created in the user’s local Temp directory, usually C:\Documents and Settings\username\Local Settings\Temp.

>The default name of the generated log file is bitrock_installer.log but if the file already exists from a previous installation, the installer will try to create an unique filename trying bitrock_installer_[pid].log and bitrock_installer_[pid]_[uid].log where [pid] is the PID of the process and [uid] is an unique identifier.

Maybe in there we can find some useful informations

delucadana, 2021-06-09:
Just getting back to this.  I had abandoned the Mac because of this intractable problem and moved to the PC where things work. Now I want to use the Mac and the online IDE because the IDE on the PC is an App and not easily accessible from other machines.
Even the online IDE, though requires the Arduino Create Agent to be running.  
The CleanMyMac people suggested removing their App and installing the Arduino Create Agent.  Tried that but it still failed.  They said, "In case you removed CleanMyMac X and still could not install the plugin, CleanMyMac X did not influence the process. "   I guess that makes sense if CleanMyMac was the culprit preventing the installation.  But if CleanMyMac cleaned out some System tool that is required for plug-in installation, then removing CleanMyMac is not going to help.  Only restoring the missing tool will help.  But that theory isn't supported either if in fact the Arduino Create Agent is, as umbynos said, a self-contained installer that doesn't require anything from the host system.  

I did a search for "bitrock" and found only one file called Isregister-501.txt created on Mar 3, 2021 which is a big 5.5MB file 3051-page document containing 24 occurrences of "bitrock" all of them related to the Bitrock InstallBuilder install or uninstall of ArduinoCreateAgent-1.1.   The file is dated Mar 3, 2021 and is located in the Downloads folder in another folder called sysdiagnose_2021.03.03_19-28-16-0500_Mac_OS_X_iMac12-2_17G14033.    Although the file creation date is in 2021, the installs all are dated 3-23-2020.  

There is no log file or any file containing mention of ArduinoCreateAgent or bitrock other than this one .txt file and the disk image (.dmg) files of the installer for the ArduinoCreateAgent.

umbynos, 2021-06-09:
I should have told you that obviously the log is generated when you run the Arduino Create Agent binary. And since it's created in `/tmp` it will be removed when the system is rebooted.

delucadana, 2021-06-09:
The ArduinoCreateAgent doesn't install so it never runs.