arduino/arduino-create-agent #637: Installation fails on Apple M1 Big Sur: post-install

ubidefeo, 2021-05-01:
## Bug Report

### Describe the bug
On Mac OS Big Sur M1 the user is requested to enter the password twice, but the second one raises a failure message
```
Problem running post-install step. Installation may not complete correctly
Error running osascript -e "do shell script \"security add-trusted-cert -d -r trustRoot -k /Library/Keychains/System.keychain /Users/ubi/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem\" with administrator privileges": 0:220: execution error: SecTrustSettingsSetTrustSettings: The authorization was denied since no user 
```
### To Reproduce
Install Create Agent on M1 Mac running Big Sur

### Expected behavior
No failure to add the certificate

## Environment (please complete the following information):
 - OS [e.g. Windows]: Mac OS
 - OS version: Big Sur 11.3
 - Browser [e.g. chrome, safari]: n/a
 - Browser Version: n/a
 - Agent Version [e.g. 1.0.0]: 1.2.3
 - CPU architecture [e.g. amd64]: arm64 (Apple M1)

## Additional context
I have manually ran the script and it fails unless I launch it with `sudo`, at which point I am prompted with the password dialog and it installs correctly

Fail
```
osascript -e "do shell script \"security add-trusted-cert -d -r trustRoot -k /Library/Keychains/System.keychain /Users/ubi/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem\""
0:190: execution error: SecCertificateAddToKeychain: Write permissions error. (1)
```

Success
```
sudo osascript -e "do shell script \"security add-trusted-cert -d -r trustRoot -k /Library/Keychains/System.keychain /Users/ubi/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem\""
Password:
```

umbynos, 2021-05-24:
It should be solved in the latest version (21.3.0) of installbuilder ([changelog](https://installbuilder.com/changelog.html))
@silvanocerza tested it and everything went smooth :butter:

umbynos, 2021-07-27:
Apparently, Silvano tested the firefox version of the agent, that is not affected by the problem, because it does not install the certificate. After another round of tests with Ubi we discovered some interesting things:
The issue appears to be the same as: https://github.com/jorangreef/sudo-prompt/issues/137

ubidefeo, 2021-11-27:
@umbynos I still have this issue.
I had to manually install the certificates and that created other issues with the IDE launching, requiring me to enter a password at every launch.
@rsora , we need this to be prioritised, I'll do a screen capture of the installation process to add here

ubidefeo, 2021-11-29:
Upon further investigation this is something we can't seem to be able to tackle unless InstallBuilder allows shell scripts to be executed.
There is a chain of trust which fails when launching an AppleScript invoking a Shell script.

For now, users who get this issue, please run this command in the terminal

`sudo security add-trusted-cert -d -r trustRoot -k /Library/Keychains/System.keychain /Users/$(whoami)/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem`

closed 2021-12-13 by umbynos