arduino/arduino-create-agent #643: Editor related not agent.

BallscrewBob, 2021-05-24:
## Bug Report

### Describe the bug
Error 522 Ray ID: 654310c6bcbc3fcd • 2021-05-24 02:24:03 UTC connection timeout.

### To Reproduce
Not sure if this is ALL plans or just subscribed plans.

### Expected behavior
open a page fully without timing out.

## Environment (please complete the following information):
 - win 10 pro
 - latest
 - chrome
 - Browser Version:
 - 1.2.3-b696989
 - amd64 x8

## Additional context
Click on the "upgrade plan" button.
Not showing anything after cloudflare Toronto other than the error listed above

per1234, 2021-05-24:
Thanks for taking the time to submit this report. I'm sorry to hear you're having a problem with the Arduino Web Editor.

I'm closing this issue because the correct place to report issues related to the Arduino Web Editor is the Arduino forum's **Software > Web Editor** category:
https://forum.arduino.cc/c/software/editor/134
If you haven't done so already, please make a post there. Thanks!

closed 2021-05-24 by per1234

BallscrewBob, 2021-05-24:
Bug Report
Describe the bug
Error 522 Ray ID: 654310c6bcbc3fcd • 2021-05-24 02:24:03 UTC connection timeout.

Target http://store.arduino.cc/digital/create#plans?source=webide&redirect=https://create.arduino.cc/editor/xxxxxxxxxxx/5c0c3346-cbc4-4393-8060-b80ab8bd733f

To Reproduce
Not sure if this is ALL plans or just subscribed plans.

Expected behavior
open a page fully without timing out.

Environment (please complete the following information):
win 10 pro
latest
chrome
90.0.4430.212 (Official Build) (64-bit)
1.2.3-b696989
amd64 x8
Additional context
Click on the "upgrade plan" button.
Not showing anything after cloudflare Toronto other than the error listed above