arduino/arduino-create-agent #657: libappindicator deprecated in Debian 11 (bullseye), agent won't run

mjackdk, 2021-08-16:
## Bug Report

### Describe the bug
"The deprecated libappindicator libraries are no longer provided" in Debian 11 (bullseye)", see [Chapter 5. Issues to be aware of for bullseye](https://www.debian.org/releases/bullseye/amd64/release-notes/ch-information.en.html#noteworthy-obsolete-packages). 

This leads to "error while loading shared libraries: libappindicator3.so.1: cannot open shared object file: No such file or directory" when trying to run the agent.

### To Reproduce
$ curl -O https://github.com/arduino/arduino-create-agent/releases/download/1.2.3/ArduinoCreateAgent-1.2.3-linux-amd64-installer-firefox.run
$ chmod u+x ArduinoCreateAgent-1.2.3-linux-amd64-installer-firefox.run
$ ./ArduinoCreateAgent-1.2.3-linux-amd64-installer-firefox.run

Complete GUI installation

Note output in terminal

### Expected behavior
Arduino Create Agent should run without problems.

## Environment (please complete the following information):
 - OS: Debian
 - OS version: 11 (bullseye)
 - Browser: Firefox
 - Browser Version: 91
 - Agent Version: 1.2.3
 - CPU architecture: amd64
## Additional context
No problems on Arch Linux, Fedora 34 or Ubuntu 21.04.

elias-james-howell, 2022-10-26:
Came across this while testing the agent on Debian 11. The fix requires installing a package.
```
sudo apt install libappindicator3-0.1-cil
```
Agent is working as intended after that.

juaringan, 2023-04-02:
For Debian 10:
```bash
# apt install libappindicator3-1 libindicator3-7
```
Fixes the problem

closed 2024-02-09 by umbynos