arduino/arduino-create-agent #688: Is there process for adding a site to the default origins list?

phptuts, 2022-01-29:
Hi,

Thank you so much for developing the Arduino Agent.  It's a fantastic tool!   

I want to use it with [ElectroBlocks](https://electroblocks.org), an open-source Arduino Blockly website.  Is there a process on the list of default origins to use the Arduino Agent with the site automatically?  I know it's a big ask, so no worries if the answer is no.

Thank you for your time.

Noah

umbynos, 2022-01-31:
Hi @phptuts, first of all congratulations on your project!
The allowed origins are currently listed in the [config.ini](https://github.com/arduino/arduino-create-agent/blob/e73b4bbad23e91eefef9a05dcbc063db1ce75e03/config.ini#L7) file, currently the only way to add a new one is by manually add one in the `config.ini` file unfortunately

phptuts, 2022-02-01:
Thank you!!  I appreciate that! @umbynos 

Would you be open to a pr that allowed that?  Would it be ok if it opened a web page like a debug page?

umbynos, 2022-02-03:
I would consider this issue solved since the topic of the issue has been answered to. Feel free to reopen it if you feel it hasn't. An answer to contribution question have been answered [in this other issue](https://github.com/arduino/arduino-create-agent/issues/380#issuecomment-1029036963)

closed 2022-02-03 by umbynos

phptuts, 2022-02-04:
@umbynos Thank you for your answer.  Would you be open to a pr that **user** configuring their origin list?   I would make it like the picture in the other issue.

[image]

fstasi, 2022-02-08:
Hi @phptuts, the user can *already* edit the `config.ini` file (which is located alongside the agent executable), and add the origins they want.

We can update the [relevant wiki page](https://github.com/arduino/arduino-create-agent/wiki/Advanced-usage#using-multiple-configurations) and better document how the user should add custom origins.

Having a custom menu item to configure just the origins is definitely not a viable solution as we are creating and exception for a specific configuration that can and should be set in the `config.ini` file.

Also, adding a custom web ui has been determined not to be feasible because of [this answer](https://github.com/arduino/arduino-create-agent/issues/380#issuecomment-1029036963)