arduino/arduino-create-agent #722: Cannot install on MB Air M2 with macOS Ventura

TurtleRad, 2022-08-07:
### Describe the problem

Trying to install ArduinoCreateAgent and failing.

I have tried with Safari and Chrome.

Issue is similar to #634 but the solutions do not work:

`sudo security add-trusted-cert -d -r trustRoot -k /Library/Keychains/System.keychain /Users/MyMBA/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem
***Error reading file /Users/MyMBA/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem
Error reading file /Users/MyMBA/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem `

`sudo osascript -e "do shell script \"security add-trusted-cert -d -r trustRoot -k /Library/Keychains/System.keychain /Users/MyMBA/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem\""
Password:
0:192: execution error: Error reading file /Users/MyMBA/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/ca.cert.pem (1)`

Errors that popped up during install:

"Error changing permissions to 040755 in /Users/simon/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app"
...and for directories there under.

I also tried out with v1.2.2

I'm new to all this Arduino stuff but I hope I haven't completely fallen over at the first step.

Many thanks for any help in advance.

### To reproduce

Downloading, installing and failing.

### Expected behavior

Downloading, installing and seeing the browser report success.

### Arduino Create Agent version

1.2.6

### Operating system

macOS

### Operating system version

Ventura

### Browser

Safari & Chrome

### Browser version

Latest

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

ubidefeo, 2022-08-08:
@TurtleRad 
Apple seems to want to change how they handle certificates at every major release.
We have recently been able to fix a long standing bug due to Apple not allowing the storage of certificates and it was not easy to resolve it.

Unfortunately until this OS gets out of beta we are not able to assign any resources to it.
We develop and test on final releases of operating systems because they tend to stay consistent across updates,
but having been on the Apple wagon for almost 20 years I can assure you they can also change things during the lifetime of an OS.

Please be patient, the behaviour on the latest OS might change before final release,
otherwise we'll look into it once it is stable

thank you
u.

PS: let's leave this one open for any other user bumping into it

Letrab, 2022-10-25:
You will get an notification that an application wants to update or delete other applications:

If you click "Allow" you should toggle the ArduinoCreateAgent on, press "Later", and click on "Retry". The installation will proceed an succeed. 

[image]
[image]
[image]
[image]

tmr-ai, 2022-10-26:
> You will get an notification that an application wants to update or delete other applications:
> 
> If you click "Allow" (or go to "Settings" -> "Privacy & Security" -> "App Management"), you should toggle the ArduinoCreateAgent on, press "Later", and click on "Retry". The installation will proceed an succeed.
> 
> [image] [image] [image] [image]

First of all, thanks a lot, your comment brought me to the right path! 

In my case, the solution was to add the installation program in the secure settings, after it was mounted, but before I started the installation process!

So I would suggest these steps to others for installation for chrome on Mac OS Ventura: 

1. download ArduinoCreate Agent
2. open downloaded Folder, but not Program yet
3. go to settings --> Privacy & Security --> App Management
4. select plus sign and add the dowloaded installation program
5. now run the installation program 
6. finished :)

umbynos, 2022-10-31:
We are currently investigating this issue. It seems the problem has always been there. Simply, macOS was not so picky. It must be something related to the macOS bundle.

sirkong2, 2022-11-01:
Same problem here. The solution suggested didn't work for me. The error that I'm getting talks is this
[image]
"The file was downloaded on an unknown date" I'm not sure if that's the problem or not but 'unknown date' doesn't sound right.

lewixlabs, 2022-11-18:
> You will get an notification that an application wants to update or delete other applications:
>
> If you click "Allow" (or go to "Settings" -> "Privacy & Security" -> "App Management"), you should [...]

I confirm, this workaround still doesn't work on macOS Ventura 13.0.1 (on my mac Mini M1 arm64).
Thanks for any suggestions.
Lewix

umbynos, 2022-11-25:
We have some news about the Agent issues on macOS Ventura
There is a new specific version ([1.2.7-ventura](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.7-ventura)) that works correctly with macOS Ventura.
You can find additional details in the recently updated support document: https://support.arduino.cc/hc/en-us/articles/6489572696220

I'm also working on a more permanent fix, you can monitor the status in the [MacOS Ventura Support milestone ](https://github.com/arduino/arduino-create-agent/milestone/1)

lewixlabs, 2022-11-25:
Let's try...

umbynos, 2023-06-12:
Hello there, we should have fixed all these problems in the latest release ([1.3.0](https://github.com/arduino/arduino-create-agent/releases/tag/1.3.0)).
The agent should be able to autoupdate itself, unless you have 1.2.7-ventura version. In that case, you have to manually uninstall it and install 1.3.0.
Please test it out so that we can close this one.

umbynos, 2023-06-23:
Closing as this should be solved, feel free to reopen it

closed 2023-06-23 by umbynos