arduino/arduino-create-agent #724: Preflight to /upload is blocked by Chrome v104+

mabraidot, 2022-09-14:
### Describe the problem

Starting with Chrome v104, requests to private networks will be blocked if they don't respond with the header:
`Access-Control-Allow-Private-Network: true`
Please, read about at [developer.chrome.com](https://developer.chrome.com/blog/private-network-access-preflight/#what-is-private-network-access-pna)

### To reproduce

Update Chrome to the latest.
Make a POST request to http://localhost:{AGENT_PORT}/upload

### Expected behavior

Response code: 200
Header: Access-Control-Allow-Private-Network: true

### Arduino Create Agent version

1.2.6

### Operating system

Windows

### Operating system version

11

### Browser

Chrome

### Browser version

105.0.5195.125

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

umbynos, 2022-09-19:
Thanks a lot for this heads up!

Apparently [from the rollout plan](https://developer.chrome.com/blog/private-network-access-preflight/#rollout-plan)
> In Chrome 107 at the earliest:
> ...
> - Chrome enforces that preflight requests must succeed, otherwise failing the requests.

This means that as of now this should be still working, but starting from Chrome 107 it will break.

After some investigation, I've found that the [library we use for the CORS](https://github.com/itsjamie/gin-cors) does not support this. We could switch to the more popular and maintained [gin-contrib/cors](https://github.com/gin-contrib/cors) but [it doesn't support preflight request yet](https://github.com/gin-contrib/cors/issues/81).

umbynos, 2023-06-23:
The timeline of chrome changed: https://developer.chrome.com/blog/private-network-access-update/#chrome-117
Chrome 117 is scheduled for september 2023

jub0bs, 2023-12-10:
@umbynos FWIW, [jub0bs/fcors](https://github.com/jub0bs/fcors) supports Private Network Access and [can be used in conjunction with Gin](https://github.com/jub0bs/fcors-examples/blob/main/gin/main.go).

closed 2023-12-18 by alessio-perugini