arduino/arduino-create-agent #792: IOT cloud can't find Arduino Agent at my MAC

Stan52, 2023-05-23:
Running  MAC OS Ventura 13.3.1 and Safari 16.4.
Arduino Agent 1.2.7 for Ventura installed successfully and started, but the application is blind at the upper dock see print screen
[SSH_5.pdf](https://github.com/arduino/arduino-create-agent/files/11547996/SSH_5.pdf)
 I was guessing that Agent is "energised" during the process of the my NANO IoT configuration but, finally the process device's setup ( sketch was uploaded ) - see print screen
[SSH_3.pdf](https://github.com/arduino/arduino-create-agent/files/11548064/SSH_3.pdf)
finished with information - see print screen
[SSH_1.pdf](https://github.com/arduino/arduino-create-agent/files/11548065/SSH_1.pdf).
Of course all recommended steps was done, but result was the same.
For me, it looks like the during the sketch uploading , the Arduino Agent cannot be found for unknown reasons.
Please help.

umbynos, 2023-05-24:
Hi @Stan52, no, the agent works correctly. Otherwise, the device setup wouldn't even start. The problem must be present somewhere else.. It would be greatly appreciated if you could provide the logs of what the agent is doing. To do so please follow the below steps:
1. Click on the tray-icon
2. Click "Open Debug Console"
3. Perform again a device setup
4. Copy all the output you get in the Debug Console here :smile:

Stan52, 2023-05-24:
Hi Umberto
Please find attached the Debug Console log
￼
I was trying once as the result of the NANO configuration is the same as previously.
Let me know what can I doing more.
Thank you
Stan

> 
> 
> Hi @Stan52 <https://github.com/Stan52>, no, the agent works correctly. Otherwise, the device setup wouldn't even start. The problem must be present somewhere else.. It would be greatly appreciated if you could provide the logs of what the agent is doing. To do so please follow the below steps:
> 
> Click on the tray-icon
> Click "Open Debug Console"
> Perform again a device setup
> Copy all the output you get in the Debug Console here 😄

{
  "Version": "1.2.7-ventura"
}

{
  "Commands": [
    "list",
    "open &lt;portName&gt; &lt;baud&gt; [bufferAlgorithm: ({default}, timed, timedraw)]",
    "(send, sendnobuf, sendraw) &lt;portName&gt; &lt;cmd&gt;",
    "close &lt;portName&gt;",
    "restart",
    "exit",
    "killupload",
    "downloadtool &lt;tool&gt; &lt;toolVersion: {latest}&gt; &lt;pack: {arduino}&gt; &lt;behaviour: {keep}&gt;",
    "log",
    "memorystats",
    "gc",
    "hostname",
    "version"
  ]
}

{
  "Hostname": "imacstaandowski.lan"
}

{
  "OS": "darwin"
}

downloadtool windows-drivers latest arduino keep

downloadtool bossac 1.7.0 arduino keep

downloadtool fwupdater latest arduino keep

downloadtool rp2040tools latest arduino keep

list

Network Ports:
[]

Serial Ports:
[
  {
    "Name": "/dev/cu.usbmodem146201",
    "SerialNumber": "607CA7F350304C39482E3120FF18261D",
    "IsOpen": false,
    "VendorID": "0x2341",
    "ProductID": "0x8057"
  }
]
[... 702 lines cut ...]
    "SerialNumber": "607CA7F350304C39482E3120FF18261D",
    "IsOpen": false,
    "VendorID": "0x2341",
    "ProductID": "0x8057"
  }
]

Network Ports:
[]

list

Network Ports:
[]

Serial Ports:
[
  {
    "Name": "/dev/cu.usbmodem146201",
    "SerialNumber": "607CA7F350304C39482E3120FF18261D",
    "IsOpen": false,
    "VendorID": "0x2341",
    "ProductID": "0x8057"
  }
]

Stan52, 2023-05-24:
Shure!

Is the board working with the IDE?

Stan52, 2023-06-21:
Hi.
Be informed that the problem seems to be solved.
I get the connection and was able to register my NANO.
But as the main condition- during board registration - the IDE must be off.
Actually I am working with Arduino IoT without any technical problems :)
So thanks.
Stan

closed 2023-06-21 by Stan52