arduino/arduino-create-agent #795: "Old agent installation found" error is not always triggered when expected

seaxwi, 2023-06-01:
### Describe the problem

The new DMG file in [ArduinoCreateAgent-1.3.1-osx-amd64-installer.dmg](https://github.com/arduino/arduino-create-agent/releases/download/1.3.1/ArduinoCreateAgent-1.3.1-osx-amd64-installer.dmg) is a great improvement from the old installer. But since the new installation path (`/Application/ArduinoCreateAgent.app`) is different from the older one (`/Users/sebastianwikstrom/Applications/ArduinoCreateAgent`), this is will not result in the old application being overwritten. Instead, the user has to manually remove the old application.

In some cases, the presense of an older Create Agent version can trigger this error:

[image]

## Problem 1: Assuming the default installation path

It seems like the trigger for the error is the presence of an `~/Applications/ArduinoCreateAgent` folder.

An older version of the Create Agent can be present **and running** from a different location, without triggering the error. This results in two instances of the application running:

[image]

On the other hand, the error is produced even if the `~/Applications/ArduinoCreateAgent` folder is empty. This seems unnecessary.

## Problem 2: Does not trigger if Create Agent is opened via Spotlight when a different version is running

Opening the application via Spotlight does not display the error or launch the application.

[image]

### To reproduce

Running two instances at the same time:

1. Install or move ArduinoCreateAgent.app (1.2.7) to a path that is not `~/Applications/ArduinoCreateAgent`
2. Run ArduinoCreateAgent.app (1.2.7) so it appears in the menu bar
3. If the `~/Applications/ArduinoCreateAgent` folder exists, remove it.
4. Run the new ArduinoCreateAgent.app (1.3.1)

🐛 Another instance of the application starts, with two Create Agent icons now appearing in the menu bar

Spotlight issue:

1. Have both 1.2.7 and 1.3.1 installed.
2. Run 1.2.7
3. Press <kbd>Command</kbd> + <kbd>Space</kbd> to open Spotlight
4. Search for 'ArduinoCreateAgent'
5. Open the newer version (in Applications by default)

🐛 Nothing happens

### Expected behavior

* I think the application shouldn't care about the presence of any particular file or folder. I would only expect to get an error if another Create Agent process is currently running.
* Opening the application via Spotlight should not prevent the error from being displayed.

### Arduino Create Agent version

1.3.1

### Operating system

macOS

### Operating system version

13.2.1 (22D68)

### Browser

Chrome

### Browser version

Version 113.0.5672.126 (Official Build) (arm64)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

umbynos, 2023-06-06:
Hi @seaxwi. 
Firstly, thanks a lot for the really detailed report.

## Regarding Problem 1:
> It seems like the trigger for the error is the presence of an `~/Applications/ArduinoCreateAgent` folder.

[You are absolutely right](https://github.com/arduino/arduino-create-agent/blob/c9c688d70c6f045538334b2d11e1e4842677023b/main.go#LL456C29-L456C29)

> An older version of the Create Agent can be present and running from a different location, without triggering the error. This results in two instances of the application running

Unfortunately, we have no other way to detect if the user changed the default installation location.

> On the other hand, the error is produced even if the `~/Applications/ArduinoCreateAgent` folder is empty. This seems unnecessary.

You are right, I'm preparing a fix for that.

## Regarding Problem 2:

> Running two instances at the same time:
>
> 1. Install or move ArduinoCreateAgent.app (1.2.7) to a path that is not `~/Applications/ArduinoCreateAgent`
> 2. Run ArduinoCreateAgent.app (1.2.7) so it [...]

I would say this is expected: Unfortunately, we have no other way to detect if the user changed the default installation location.

> Spotlight issue:
> 
> 1. Have both 1.2.7 and 1.3.1 installed.
> 2. Run 1.2.7
> 3. Press <kbd>Command</kbd> + <kbd>Space</kbd> to open Spotlight
> 4. Search for 'ArduinoCreateAgent'
> 5. Open the newer version (in Applications by default)
> 
> 🐛 Nothing happens

This seems strange, I'm investigating this.

> I think the application shouldn't care about the presence of any particular file or folder. I would only expect to get an error if another Create Agent process is currently running.

Unfortunately, this is not possible: https://github.com/arduino/arduino-create-agent/issues/725#issuecomment-1250836290

umbynos, 2023-06-08:
@seaxwi let me know when you have the chance to test #797

PR #797 (2023-06-09): Check the folder content in new version of the agent on macos
**Please check if the PR fulfills these requirements**

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-create-agent/pulls)
      before creating one)
- [ ] Tests for the changes have been added (for bug fixes / features)

* **What kind of change does this PR introduce?**

bug fix

- **What is the current behavior?**

when uninstalling an old version of the agent (e.g. 1.2.7) a folder in `~/Applications/ArduinoCreateAgent` could be left behind (empty, but existing), after un [... cut]
closed 2023-06-09 by umbynos