arduino/arduino-create-agent #803: Linux Ubuntu install file unable to install

milesnz, 2023-07-04:
### Describe the problem

When attempting to run the linux binary RUN file nothing happens. It does nothing.
I tried downloading the run files and also the zip and the tz file and none of them run.
Permissions are correct but nothing happens.

### To reproduce

Download the run file ArduinoCreateAgent-1.3.2-linux-amd64-installer.run
Double click the file in FileManager - nothing happens
Open folder in terminal and then try run using ./ArduinoCreateAgent-1.3.2-linux-amd64-installer.run - get error message.

> ls -al ArduinoCreateAgent-1.3.2-linux-amd64-installer.run 
-rwxrwxr-x 1 miles miles 25827027 Jul  4 19:54 ArduinoCreateAgent-1.3.2-linux-amd64-installer.run

./ArduinoCreateAgent-1.3.2-linux-amd64-installer.run 
bash: ./ArduinoCreateAgent-1.3.2-linux-amd64-installer.run: No such file or directory

### Expected behavior

Either methods should install the Agent but nothing happens or the error appears in the CLI

### Arduino Create Agent version

1.3.2 linux

### Operating system

Linux

### Operating system version

Ubuntu 18.04, Ubuntu 20.04, Ubuntu 22.04

### Browser

Chrome and Firefox

### Browser version

Chrome 114.0.5735.198 (Official Build) (64-bit),  Firefox 114.0.2-1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

umbynos, 2023-07-04:
You are definitely right, and thanks for reporting it.
It seems that for some reason, the installers for linux are built for 32-bit linux instead of 64-bit. :fearful: 
This happened before 1.3.0 release:

```
$ file ArduinoCreateAgent-1.3.0-linux-amd64-installer.run 
ArduinoCreateAgent-1.3.0-linux-amd64-installer.run: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.2.5, stripped
$ file ArduinoCreateAgent-1.2.7-linux-amd64-installer-chrome.run 
ArduinoCreateAgent-1.2.7-linux-amd64-installer-chrome.run: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.4.0, stripped
```
Currently, the affected releases are:
- [1.3.0](https://github.com/arduino/arduino-create-agent/releases/tag/1.3.0)
- [1.3.1](https://github.com/arduino/arduino-create-agent/releases/tag/1.3.1)
- [1.3.2](https://github.com/arduino/arduino-create-agent/releases/tag/1.3.2)

I suspect that this has something to do with changes that we did in the installbuilder configuration file.
I know this is not ideal, but while we fix the bug you can install the latest working version which is [1.2.7](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.7). The agent should be able to autoupdate to the latest (1.3.2) automatically when you connect to [Arduino Web Editor](https://create.arduino.cc/editor/).

milesnz, 2023-07-05:
Thank you for such a fast response. The agent installs now and runs automatically on the 1.3.2 version.
The agent finds my online esp8266 devices but doesn't see my USB attached ESP32-CAM - I'll need to find how out how to attach my usb devices. /dev/tty shows no USB devices either
Thanks for your help

jfclere, 2023-08-28:
[jfclere@dhcp-144-158 pico]$ file ArduinoCreateAgent-1.3.2-linux-amd64-installer.run
ArduinoCreateAgent-1.3.2-linux-amd64-installer.run: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.2.5, stripped

PR #847 (2023-10-25): fix agent being built for 32 bit linux for real
**Please check if the PR fulfills these requirements**

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-create-agent/pulls)
      before creating one)
- [ ] Tests for the changes have been added (for bug fixes / features)

* **What kind of change does this PR introduce?**

bugfix :lady_beetle: 
- **What is the current behavior?**

The agent builds after 1.2.7 on linux have different kind of problems:
1. `1.3.0`, `1.3.1`, `1.3.2` are being built for linux 32 bit (see #803)
2. `1.3.3` does  [... cut]
closed 2023-09-08 by umbynos