arduino/arduino-create-agent #806: Cannot uninstall Arduino Create Agent 1.1

Coder-256, 2023-07-13:
### Describe the problem

I have a very old copy of Arduino Create Agent installed in `~/Applications/ArduinoCreateAgent-1.1`. It's a 32-bit Intel binary so I cannot even run the uninstaller, and newer versions no longer come with a compatible uninstaller. Is there a list of files I can manually remove?

### To reproduce

N/A

### Expected behavior

N/A

### Arduino Create Agent version

N/A but I version 1.3.2 doesn't have an uninstaller

### Operating system

macOS

### Operating system version

13.4.1 (c)

### Browser

Chrome

### Browser version

114.0.5735.198

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

Coder-256, 2023-07-13:
Actually for context, I'm trying to uninstall because I keep getting crashes like this every time I log in:

<details>
<summary><code>Arduino_Create_Bridge-2023-07-12-212402.ips</code></summary>
	-------------------------------------
	Translated Report (Full Report Below)
	-------------------------------------
	
	Process:               Arduino_Create_Bridge [881]
	Path:                  /Users/USER/*/ArduinoCreateAgent.app/Contents/MacOS/Arduino_Create_Bridge
	Identifier:            create.arduino.cc
	Version:               0.1 (0.1)
	Code Type:             X86-64 (Translated)
	Parent Process:        launchd [1]
	User ID:               501
	
	Date/Time:             2023-07-12 21:23:55.0632 -0400
	OS Version:            macOS 13.4.1 (22F770820d)
	Report Version:        12
	Anonymous UUID:        52BB50D0-D67A-976B-88FE-73A57085AC44
	
	
	Time Awake Since Boot: 47 seconds
	
	System Integrity Protection: enabled
	
	Crashed Thread:        0  Dispatch queue: com.apple.main-thread
	
	Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
	Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000000000f1
	Exception Codes:       0x0000000000000001, 0x00000000000000f1
	
	Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
	Terminating Process:   exc handler [881]
	
	VM Region Info: 0xf1 is not in any region.  Bytes before following region: 140723139690255
	      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
	      UNUSED SPACE AT START
	--->  
	      mapped file              7ffca8c0c000-7ffcb7d40000 [241.2M] r-x/r-x SM=COW  ...t_id=25aa5a7d
	
	Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
	0   <translation info unavailable>	       0x104dcbbe0 ???
	1   <translation info unavailable>	       0x104dcbbd4 ???
	2   Arduino_Create_Bridge         	         0x4040f03 runtime.signalstack + 99
	3   Arduino_Create_Bridge         	         0x4040c54 runtime.minitSignalStack + 132
	4   Arduino_Create_Bridge         	         0x4029660 runtime.minit + 32
	5   Arduino_Create_Bridge         	         0x40300ec runtime.mstart1 + 108
	6   Arduino_Create_Bridge         	         0x4030074 runtime.mstart + 100
	7   Arduino_Create_Bridge         	         0x4053ea9 runtime.rt0_go + 409
	
	Thread 1:: com.apple.rosetta.exceptionserver
	0   runtime                       	    0x7ff7ffd9d694 0x7ff7ffd99000 + 18068
	
	
	Thread 0 crashed with X86 Thread State (64-bit):
	  rax: 0x000000000000000c  rbx: 0x0000000004904540  rcx: 0x0000000208cea818  rdx: 0x0000000000000000
	  rdi: 0x0000000208cea818  rsi: 0x0000000000000000  rbp: 0x0000000208cea830  rsp: 0x0000000208cea800
	   r8: 0x0000000000000001   r9: 0x0000000000000008  r10: 0x0000000000000010  r11: 0x000000c420010020
	  r12: 0x0000000000000001  r13: 0x0000000000000002  r14: 0x0000000000000008  r15: 0x0000000000000400
	  rip: <unavailable>       rfl: 0x0000000000000243
	 tmp0: 0x00000000000000f1 tmp1: 0x00000000000000f1 tmp2: 0x0000000004040f03
	
	
	Binary Images:
	       0x104beb000 -        0x104c86fff dyld (*) <5db85b72-c63a-3182-91e5-5c942ec30e48> /usr/lib/dyld
	    0x7ff7ffd99000 -     0x7ff7ffdc8fff runtime (*) <c4d322c3-30c5-3c52-adb8-2fd12ec92bc0> /usr/libexec/rosetta/runtime
	       0x10000a000 -        0x10005dfff libRosettaRuntime (*) <73cd5d0d-71d4-353e-9740-af3964d9a4ae> /Library/Apple/*/libRosettaRuntime
	         0x4000000 -          0x48befff create.arduino.cc (0.1) <51c00d6a-b529-303f-9e9b-dbfc1b336b7e> /Users/USER/*/ArduinoCreateAgent.app/Contents/MacOS/Arduino_Create_Bridge
	               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???
[... 159 lines cut ...]
	},
	  "logWritingSignature" : "09264f5c37456386e5e650cd4ecf47ee10425826",
	  "trialInfo" : {
	  "rollouts" : [
	    {
	      "rolloutId" : "5fb4245a1bbfe8005e33a1e1",
	      "factorPackIds" : {
	
	      },
	      "deploymentId" : 240000021
	    },
	    {
	      "rolloutId" : "60da5e84ab0ca017dace9abf",
	      "factorPackIds" : {
	
	      },
	      "deploymentId" : 240000008
	    }
	  ],
	  "experiments" : [
	
	  ]
	}
	}
</details>

umbynos, 2023-08-23:
It should be sufficient to remove `~/Applications/ArduinoCreateAgent-1.1` folder

closed 2023-09-19 by Coder-256