arduino/arduino-create-agent #816: on fedora f38 the ArduinoCreateAgent-1.3.0-linux-amd64-installer.run doesn't work.

jfclere, 2023-08-27:
### Describe the problem

I have downloaded in $HOME/pico
ArduinoCreateAgent-1.3.0-linux-amd64-installer.tar.gz
extracted it and run it and get:
[jfclere@2a02-842b-8452-dd01-3aa1-1696-2cfa-2836 ~]$ pico/ArduinoCreateAgent-1.3.0-linux-amd64-installer.run
bash: pico/ArduinoCreateAgent-1.3.0-linux-amd64-installer.run: cannot execute: required file not found

### To reproduce

As in describe.
Note:
[jfclere@2a02-842b-8452-dd01-3aa1-1696-2cfa-2836 pico]$ strace ./ArduinoCreateAgent-1.3.0-linux-amd64-installer.run
execve("./ArduinoCreateAgent-1.3.0-linux-amd64-installer.run", ["./ArduinoCreateAgent-1.3.0-linux"...], 0x7fff7cdfdb10 /* 65 vars */) = -1 ENOENT (No such file or directory)
strace: exec: No such file or directory
+++ exited with 1 +++

### Expected behavior

no error message and agent  is installed.

### Arduino Create Agent version

1.3.0

### Operating system

Linux

### Operating system version

fedora 38

### Browser

firefox

### Browser version

116.0.3 (64-bit)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

umbynos, 2023-08-28:
Should be a duplicate of #803. Could you please try to verify if installing 1.2.7 works?

jfclere, 2023-08-28:
Yes it does.
Closing as duplicate of https://github.com/arduino/arduino-create-agent/issues/803

closed 2023-08-28 by jfclere