arduino/arduino-create-agent #845: Unable to install the Agent on Debian 12 (required file not found)

yves84, 2023-10-10:
### Describe the problem

Unable to install the agent under Debian 12 (fresh install, it would be nice to know witch file is required).

### To reproduce

Simple try to install:
```
user@L420-1:~/Downloads$ chmod +x ArduinoCreateAgent-1.3.2-linux-amd64-installer.run 
user@L420-1:~/Downloads$ ./ArduinoCreateAgent-1.3.2-linux-amd64-installer.run 
bash: ./ArduinoCreateAgent-1.3.2-linux-amd64-installer.run: cannot execute: required file not found
```

### Expected behavior

Opening GUI or somthing ...

### Arduino Create Agent version

1.3.2

### Operating system

Linux

### Operating system version

Debian 12

### Browser

Firefox

### Browser version

115.3.0esr

### Additional context

```
user@L420-1:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

user@L420-1:~$ uname -a
Linux L420-1 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux
```

Desktop: Cinnamon 5.6.8
bash 5.2.15

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

umbynos, 2023-10-10:
Hello! Could you please run `file` on that downloaded binary? It should be a duplicate of https://github.com/arduino/arduino-create-agent/issues/803.
It's already fixed on main, but we need to make a release (it will happen in the next days).
As a workaround for now you can install the latest working version which is [1.2.7](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.7). The agent should be able to autoupdate to the latest (1.3.2) automatically when you connect to [Arduino Web Editor](https://create.arduino.cc/editor/).

yves84, 2023-10-11:
Sure

```
user@L420-1:~/Downloads$ file ArduinoCreateAgent-1.3.2-linux-amd64-installer.run 
ArduinoCreateAgent-1.3.2-linux-amd64-installer.run: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.2.5, stripped
```
Yes, seems like #803 but with another error msg.

ty!

umbynos, 2023-10-11:
Perfect, closing as duplicate then :smile:

closed 2023-10-11 by umbynos