arduino/arduino-create-agent #846: Create Agent v1.3.3 Linux installers do not contain Arduino_Create_Agent executable

greggwardnz, 2023-10-19:
### Describe the problem

The ArduinoCreateAgent-1.3.3-linux-amd64-installer.run and ArduinoCreateAgent-1.3.3-linux-amd64-installer.tar.gz installation packages for Arduino Create Agent, do not contain the Arduino_Create_Agent executable. The file sizes are 4.64 MB and 3.99 MB, respectively, compared to the v1.3.2 version of the Linux installation packages, which are 24.6 MB for the ArduinoCreateAgent-1.3.2-linux-amd64-installer.run file and 23.9 MB for the ArduinoCreateAgent-1.3.2-linux-amd64-installer.tar.gz

When running the ArduinoCreateAgent-1.3.3-linux-amd64-installer.run installer, it outputs the following during the install and no Arduino_Create_Agent executable is available in the install path:

Please wait while Setup installs arduino-create-agent on your computer.

 Installing
 0% ______________ 50% ______________ 100%
 ###########################/bin/sh: 1: /home/greggw/ArduinoCreateAgent/Arduino_Create_Agent: not found
##############

----------------------------------------------------------------------------
Setup has finished installing arduino-create-agent on your computer.

### To reproduce

Run ArduinoCreateAgent-1.3.3-linux-amd64-installer.run and accept the default options.

### Expected behavior

Running ArduinoCreateAgent-1.3.3-linux-amd64-installer.run installer would write a Arduino_Create_Agent,, Arduino_Create_Agent-cli and Arduino_Create_Agent-temp executable into the installation path.

### Arduino Create Agent version

v1.3.3

### Operating system

Linux

### Operating system version

Ubuntu 23.04

### Browser

Brave

### Browser version

v1.59.117

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

umbynos, 2023-10-23:
Could you please try again[ with 1.3.3](https://github.com/arduino/arduino-create-agent/releases/tag/1.3.3)? I replaced the release binaries with manually generated ones. I spotted the problem in the CI but I still have to fix it

umbynos, 2023-10-23:
I pinpointed the problem to this commit https://github.com/arduino/arduino-create-agent/pull/778/commits/b62a2ff745c804f12ea1dcb6d8856602d5aa8545#diff-87db21a973eed4fef5f32b267aa60fcee5cbdf03c67fafdc2a9b553bb0b15f34R426

greggwardnz, 2023-10-23:
The issue is now resolved with the latest v1.3.3 release binaries. Thank you.

PR #847 (2023-10-25): fix agent being built for 32 bit linux for real
**Please check if the PR fulfills these requirements**

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-create-agent/pulls)
      before creating one)
- [ ] Tests for the changes have been added (for bug fixes / features)

* **What kind of change does this PR introduce?**

bugfix :lady_beetle: 
- **What is the current behavior?**

The agent builds after 1.2.7 on linux have different kind of problems:
1. `1.3.0`, `1.3.1`, `1.3.2` are being built for linux 32 bit (see #803)
2. `1.3.3` does  [... cut]
closed 2023-10-23 by per1234