arduino/arduino-create-agent #856: The "Installing drivers" pop-up window keeps appearing constantly.

Jorgetrujilloroman, 2023-11-20:
### Describe the problem

I have been facing a persistent bug with the Create Agent driver for some time now. Whenever I open an Arduino Cloud tab on my browser, a pop-up window for driver installation appears randomly. Clicking either 'Yes' or 'No' doesn't seem to do anything but more pop-ups keep appearing while the IoT Cloud tab remains open.

I have already installed the latest Create Agent release 1.3.5. 

I noticed that other users have been facing the same issue, as shown [in this Forum post](https://forum.arduino.cc/t/drivers-popups-and-more-popups/1178392).

[image]

### To reproduce

It is a bit difficult to reproduce the problem since, for example, in my case, I have a laptop and a fixed PC, both with Windows 10 and the same version of Arduino Create Agent (1.3.5) installed but the problem only occurs on my laptop.

### Expected behavior

The problem is that even if I click "yes" or "no", nothing happens and the driver installation window keeps popping up infinitely as long as I keep a Web Editor or IoT Cloud tab open.

### Arduino Create Agent version

1.3.5

### Operating system

Windows

### Operating system version

10 Home

### Browser

Google Chrome

### Browser version

119.0.6045.160 (Build oficial) (64 bits)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

per1234, 2023-11-24:
I have [the `crashreport` key set to `true` in my configuration file](https://github.com/arduino/arduino-create-agent/wiki/How-to-use-crash-report-functionality). This log file was produced when the fault occurred on my system:

```text
Failed to write to log, write /dev/stdout: The handle is invalid.
time="2023-11-24T12:22:24-08:00" level=info msg=" id=HKr35cHt req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:22:24-08:00" level=info msg=" id=HKr35cHt status=200 bytes=0 time=395.4µs"
time="2023-11-24T12:22:24-08:00" level=info msg=" id=HVRAndQo req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:22:24-08:00" level=info msg=" id=HVRAndQo status=200 bytes=0 time=234.4µs"
time="2023-11-24T12:22:25-08:00" level=info msg=" id=u9lceJls req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:22:25-08:00" level=info msg=" id=u9lceJls status=200 bytes=0 time=12.0321ms"
time="2023-11-24T12:22:45-08:00" level=info msg=" id=PW64NhV5 req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:22:45-08:00" level=info msg=" id=PW64NhV5 status=200 bytes=0 time=92.9µs"
time="2023-11-24T12:24:19-08:00" level=info msg=" id=9fpTgyJo req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:24:19-08:00" level=info msg=" id=9fpTgyJo status=200 bytes=0 time=0s"
time="2023-11-24T12:25:19-08:00" level=info msg=" id=1swbBLJc req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:25:19-08:00" level=info msg=" id=1swbBLJc status=200 bytes=0 time=0s"
time="2023-11-24T12:26:19-08:00" level=info msg=" id=KMC7DKe3 req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:26:19-08:00" level=info msg=" id=KMC7DKe3 status=200 bytes=0 time=0s"
time="2023-11-24T12:26:36-08:00" level=info msg=" id=O8kDrohB req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:26:36-08:00" level=info msg=" id=O8kDrohB status=200 bytes=0 time=188.2µs"
time="2023-11-24T12:28:14-08:00" level=info msg=" id=zyPuTHG5 req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:28:14-08:00" level=info msg=" id=zyPuTHG5 status=200 bytes=0 time=0s"
time="2023-11-24T12:29:14-08:00" level=info msg=" id=ROa3EH8O req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:29:14-08:00" level=info msg=" id=ROa3EH8O status=200 bytes=0 time=139.9µs"
time="2023-11-24T12:30:14-08:00" level=info msg=" id=-lTcIJvt req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:30:14-08:00" level=info msg=" id=-lTcIJvt status=200 bytes=0 time=369.2µs"
time="2023-11-24T12:30:37-08:00" level=info msg=" id=b6D20iYv req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:30:37-08:00" level=info msg=" id=b6D20iYv status=200 bytes=0 time=500.4µs"
time="2023-11-24T12:31:15-08:00" level=info msg=" id=SYlrUXMG req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:31:15-08:00" level=info msg=" id=SYlrUXMG status=200 bytes=0 time=0s"
time="2023-11-24T12:32:15-08:00" level=info msg=" id=82udJ-2_ req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:32:15-08:00" level=info msg=" id=82udJ-2_ status=200 bytes=0 time=0s"
time="2023-11-24T12:33:15-08:00" level=info msg=" id=1cciug88 req=HEAD /v2/pkgs/tools/installed from=127.0.0.1"
time="2023-11-24T12:33:15-08:00" level=info msg=" id=1cciug88 status=200 bytes=0 time=63.6µs"
```

The first line `Failed to write to log, write /dev/stdout: The handle is invalid.` is always produced when you start Arduino Create Agent on a Windows system due to it not having access to output streams because it is [built as a "GUI binary"](https://github.com/arduino/arduino-create-agent/blob/d4c453787561de7886d9224c97b315ea1185f89e/Taskfile.yml#L50). So I don't think that line is relevant. But the 15 entries related to `/v2/pkgs/tools/installed` seem correlated with the 14 (yes, it is one less than the number of log entries) instances of the "**Installing drivers**" dialog that appeared unexpectedly on my computer.

ldw300, 2023-12-02:
I am having the same issue, the first time it happened it spammed like crazy. However, every time it does it now, it only opens 2 or 3 windows. I have tried uninstalling all Arduino IDE's and the create agent and reinstalling them. Same issue. Als o restarted computer and verified the COM5 port my Uno r4 is connected to is open and not stuck. 

[image]

Also clicking "yes" or "no" has no effect.

jpgreenh, 2024-01-01:
This issue is driving me crazy!! Why can't a fix to this driver pop-up matter be provided??

umbynos, 2024-03-26:
We just released version 1.4.0, does the issue is still valid?

umbynos, 2024-03-26:
Fantastic! I would close this one then. Feel free to reopen the issue if this happens again

closed 2024-03-26 by umbynos

Jorgetrujilloroman, 2024-03-27:
> I haven't experienced it since #897
> 
> I have been monitoring the forum for user reports of this problem and the last one I saw was 2024-02-17, which matches well with the release of that fix in 1.3.9.

I haven't experienced it yet either :)