arduino/arduino-create-agent #863: Invalid Signature Error with .NET Core and Arduino

erhankiyakk, 2023-12-05:
### Describe the problem

I am encountering an "invalid signature" error while working with .NET Core and Arduino. Here's how I am performing the signing process:

`private string SignCode()
{
    try
    {
        // Command line string with AVRDUDE configuration
        string commandLine = "\"{runtime.tools.avrdude.path}/bin/avrdude\" \"-C{runtime.tools.avrdude.path}/etc/avrdude.conf\" -v  -patmega328p -carduino \"-P{serial.port}\" -b115200 -D \"-Uflash:w:{build.path}/{build.project_name}.hex:i\"";

        // Private key in PEM format
        string privateKeyPem = @"-----BEGIN RSA PRIVATE KEY-----
MIIEogIBAAKCAQEAyh9OfQnmto6A7ext/c2+mszLlmhiA7SnhPlBhrecvpaQB3mH
zp1e8B01S1sCC1HuGphDP4yRmJJK/EwWDlPrqKZg5xwF9N7nam83djPyY6hkH1NJ
boBe6Zu9vyOV79YpxxPaYDRmgXG5u46OCjVZctdHGWC9LDOzVeDlPTkgoqG4lRSQ
NRFlkc9cNh7BZt1oj/Th/ZEMhFTDcSlZTx0F8nz1J6FrLH1lt6JfHxBrvzTk89pu
kolkm8D6X+W0eEm+zJsk0fSGfvLpu+De28BFHrxvLSrguVJUzZalqjqhm2hFP6uh
0gnmb8LeKiDLdGkySPoY0NcaJiLNmJrVEj8rNwIDAQABAoIBAF1eLOGx6zO1jHVb
3Et7+WqE5B7sxSo1D5VrlwtmUi/wWUQ6rbOSGsMMmwveVIlqJXV/lMEMbE1wdtcp
Rp8DwT687NKVrWK6fWg7TAF8aaIDrYyS6r7y6eU8HmDnTWHnKpFd6Gi8jAWLxWKQ
railkFZuf9eHSpvaniMzqnGnrB2zh81wXvwP3/bImdo/hRP8r8V48HCOlEOhMWAj
SewAcUkHmVSSMQqqLHrneAOpdiBneq281l2EEeKJl58R9OD/jx36Tc5Oyi0i2ucc
x2n5cU653B3Eib4dKDRratuP7G289Pv74+fZreFUqmvVK6tj/08ftcsnUXy/eEIL
9i3vhtECgYEA/Bw93ioxW6Xpedyds6yHQfhkE/0rnAvhjZsRqrNCii08LHLPd9fo
xLvKkqEdlBkZc+i1RzMM6YsuRWKi/m9P8vvqK5cpOkSnKdtozUNHeNHBdKczydzP
DGt0GF5drUDPUmI0KkpupiJFNftXFkbNpYc9P3EiGZrYXOKWaC32O+UCgYEAzT2g
qEwhz2H9uZnX657BNP/5vuLki2SZhlxnUjTfOZSj0Z4cb2Qc1H09GFm+d5JfvaVA
0kA6lDVCk1y7yTSWsI6AeJ/FmCFbzIKxjIHRtilE07rYF+eIL0FhFRwlrVNUlV3S
qI91DH35ODhDH+gER/t3ovoW0R8puTwPwS7acOsCgYBUDLJPk+96p3iMKRHmadl6
E+6Bf4D/eDKNPCqKp1LCsEwEYY5rXO78T8M8WTx8OFX0/5sYDpv6bG1Tsou2mbOk
xXmPaMjU5Th5yu1IqWp1jDCbzHaGkPfyIwzEZwmCDj5QL+9GPiRffMNoONAS3C4J
RmoT8Yq10JrbdKtCNjiGRQKBgF6zsQ33EyK99HVPp3HxLvVlp7eMuigXoz98I7Cq
9NV84Jo4eqZra2Ig56/0uQ4zXDoFpoNdX6pDaRTWFVAiE+Yj2uPN1T79JX6YT5Sl
0j5bjIamEE/3Zb9oS4OSGVCAwuapBCsDBt9cwUJPs6llRZRxVCK/gwe7eWUfcaWS
GPlBAoGAVIcM+d4x8NBOSYzBrUWazj2+5eiB8v1SvIMonm0zpeGfTByxoK3CGvTB
ECz65EZrzBVHb1QYLV0ZUK17azsO7r3Y3rL7PCj6t2ADIUwdO0TtRO5Icy37i/to
uEnKrszlB/alkWVhkfKaq9GGFhfQkdsAj+0eMALy1XUA9ksUV0Y=
-----END RSA PRIVATE KEY-----";

        using (RSA rsa = RSA.Create())
        {
            rsa.ImportFromPem(privateKeyPem.ToCharArray());

            using (SHA256 sha256 = SHA256.Create())
            {
                byte[] dataBytes = Encoding.UTF8.GetBytes(commandLine);
                byte[] hash = sha256.ComputeHash(dataBytes);
                byte[] signatureBytes = rsa.SignHash(hash, HashAlgorithmName.SHA256, RSASignaturePadding.Pkcs1);
                return Convert.ToBase64String(signatureBytes);
            }
        }
    }
    catch (Exception ex)
    {
        return ex.ToString();
    }
}
`

Additionally, I've added the following to the config.ini file:
`signatureKey = -----BEGIN PUBLIC KEY-----\nMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAyh9OfQnmto6A7ext/c2+mszLlmhiA7SnhPlBhrecvpaQB3mHzp1e8B01S1sCC1HuGphDP4yRmJJK/EwWDlPrqKZg5xwF9N7nam83djPyY6hkH1NJboBe6Zu9vyOV79YpxxPaYDRmgXG5u46OCjVZctdHGWC9LDOzVeDlPTkgoqG4lRSQNRFlkc9cNh7BZt1oj/Th/ZEMhFTDcSlZTx0F8nz1J6FrLH1lt6JfHxBrvzTk89pukolkm8D6X+W0eEm+zJsk0fSGfvLpu+De28BFHrxvLSrguVJUzZalqjqhm2hFP6uh0gnmb8LeKiDLdGkySPoY0NcaJiLNmJrVEj8rNwIDAQAB\n-----END PUBLIC KEY-----`

I'm attempting to upload the data with this configuration:
`let data = {
    "port": port,
    "data": res.Hex,
    "hex": res.Hex,
    "filename": res.Filename,
    "network": res.Network,
    "commandline": res.Commandline,
    "board": res.Board,
    "serial": true,
    "signature": res.Signature
};
$.ajax({
    url: "http://localhost:8991/upload",
    data: JSON.stringify(data),
    success: function (t) {
        console.log(t);     
    },
    error: function (t) {
        console.log(t);
        console.log(t.responseText);
    }
});
`

However, I keep receiving an "invalid signature" error. How can I resolve this issue?

### To reproduce

1. Setup Environment: Ensure you have a .NET Core environment set up for interfacing with Arduino.
2. Add Code for Signature Generation: Implement the SignCode function in your .NET Core application as provided in the original description. This function generates a signature using RSA private key.
3. Configure config.ini: Add the public key to your config.ini file as indicated in the original description.
4. Prepare Data for Upload: Construct the data object in your code with necessary fields such as port, data, hex, filename, network, commandline, board, serial, and signature.
5. Execute AJAX Request: Perform an AJAX request to http://localhost:8991/upload with the data object.
6. Observe Error: The response received is an "invalid signature" error.

### Expected behavior

- The .NET Core application successfully generates a valid digital signature using the RSA private key.
- The data object, along with the generated signature, is sent to the server via AJAX.
- The server validates the signature against the public key provided in the config.ini file.
- Upon successful validation, the server accepts the upload without returning an "invalid signature" error.
- The entire process completes smoothly, allowing for the intended data transfer between the .NET Core application and the server.

### Arduino Create Agent version

Last Version 

### Operating system

Windows

### Operating system version

Windows 11

### Browser

Chrome

### Browser version

Last version

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

dido18, 2025-04-01:
Hi @erhankiyakk  thanks for the report.

We recently released the agent [1.7.0 ](https://github.com/arduino/arduino-create-agent/releases/tag/1.7.0) that contains a fix for the `signatureKey` format.

Could you test it with the new version?
As you can read from the updated [wiki](https://github.com/arduino/arduino-create-agent/wiki/Advanced-usage#how-to-use-a-custom-privatepublic-key) page,  the public key must be set in the `config.ini` file using this format:

```
signatureKey = `-----BEGIN PUBLIC KEY-----
MIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAtHP52M8dCYDGaC4UaOvU
cfLqhteGX75EnbXMr6iOg7r7Of+doFV+Ee233Ly/di15CXaju3EpgUka5QSu6z2m
4sne32aGw6T/eggY636CKcRHFPFjLmXX0CHq+Evg3E4g8W7Yslo2qu1SP3ySZCqe
VpZHSeehlxFPpQKWXa1YiNF0gWh3cNQ0wneOsJ+ndShSuQ5C2YnSEoeoiEGVFOS0
evX4GEdadudGBjHQUKjhj+k3Ydaz014aIIC7CUVkQog9B7vpB+znHJH/gCl9DqYO
4mjPfHG4c5ppNu455hEe75R5q9bPc7TjBA3jpZsdrBY05lX2Q2nAQgSYIHpf78xl
7wIDAQAB
-----END PUBLIC KEY-----`
```

dido18, 2025-04-01:
I close the issue.
Feel free to reopen it if the problem persists 💯

closed 2025-04-01 by dido18