arduino/arduino-create-agent #907: Improve HTTPS Certificates detection and installation on macOS (Safari)

Xayton, 2024-02-28:
### Describe the request

Many users reported that the Agent does not work on macOS using Safari. Most of the time this is because they have not installed the HTTPS Certificates for localhost (this is needed only with Safari, due to the way the browser handles connections to localhost).

The HTTPS Certificates installation should be made more robust and automatic, while keeping it optional and only done when really needed (only if using Safari, and not other browsers).

- Add a function to check if the certificates are already present and get also when they will expire
- Add a function to remove existing certificates (expired or not)
- Add a way to prompt the user for confirmation (like a dialog window)
- Add a new setting `installCertificates` used to save if the user want's to install them or not (true, false, missing). If missing, the user should be prompted at the first check.
- Add some logic at startup, that will prompt the user if they want to install the HTTPS Certificates, explaining it's needed if they want to use the Agent with Safari.
- Add some logic so that the check above can be scheduled once every X time, and not just at startup. This way it will work even if the user keeps the Agent running for days. This function could be used in the future for other reasons too, like an auto-update functionality.

### Describe the current behavior

N/A

### Arduino Create Agent version

1.3.9

### Operating system

macOS

### Operating system version

N/A

### Browser

Safari

### Browser version

N/A

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the feature was still missing when using the latest version
- [X] My request contains all necessary details

Xayton, 2024-03-15:
It would be nice to automate the certificate detection so that it happens only when needed and only if the user is on Safari. One possible way would be to trigger it if the agent receives a request with Safari user-agent, even if that request would be then blocked by the browser CORS or other protections.  
Not sure if this works though, need some investigation, because the localhost HTTPs requirement of Safari might block completely the call to the agent.

cmaglie, 2024-03-15:
My forecast for the above activities is about 1 week, BTW I'm wondering if displaying a dialog to **all** users by default is the right thing to do since we may use the browser user-agent to detect when the user is running safari and show a warning directly on the browser if the connection to the agent fails **and** the browser is safari. This could be something to discuss in detail with the design team.

Anyway, here are the details of the research:

> Add a function to check if the certificates are already present and get also when they will expire

I found a couple of ways to do this you need the "certificate" and the "policy" applied to that certificate (the policy is the set of rules that the certificate should satisfy to be considered valid) and check if the [policy is matched](https://support.apple.com/en-il/guide/keychain-access/kyca2794/mac#:~:text=In%20the%20Keychain%20Access%20app,%3E%20Evaluate%20%5Bcertificate%20name%5D.).
This can be done with a CLI tool called [`security`](https://superuser.com/a/1700716/637881) or programmatically using the security API:
- [Security API overwire](https://developer.apple.com/documentation/security?language=objc) 
- [Cert and Trust services](https://developer.apple.com/documentation/security/certificate_key_and_trust_services/trust?language=objc)
- [Creating a Trust object](https://developer.apple.com/documentation/security/certificate_key_and_trust_services/trust/creating_a_trust_object?language=objc)
- [Evaluate a Trust](https://developer.apple.com/documentation/security/certificate_key_and_trust_services/trust/evaluating_a_trust_and_parsing_the_result?language=objc)
- [Discover why a Trust evaluation failed](https://developer.apple.com/documentation/security/certificate_key_and_trust_services/trust/discovering_why_a_trust_evaluation_failed?language=objc)

Forecast for this activity: 1...2 days

> Add a function to remove existing certificates (expired or not)

This should use the [API we already use to install](https://github.com/arduino/arduino-create-agent/blob/main/certificates/install_darwin.go#L39) them, I think the required methods are:
* [Searching for keychain items](https://developer.apple.com/documentation/security/keychain_services/keychain_items/searching_for_keychain_items?language=objc)
* [SecItemDelete](https://developer.apple.com/documentation/security/1395547-secitemdelete?language=objc)

Forecast for this activity: 1...2 days

> Add a way to prompt the user for confirmation (like a dialog window)
> Add a new setting installCertificates used to save if the user want's to install them or not (true, false, missing). If missing, [...]

This is easy, we are [already using `osascript`](https://github.com/arduino/arduino-create-agent/blob/main/certificates/install_darwin.go#L85-L87) to show dialogs the rest are just plain checks.

Forecast for this activity: 1 day

> Add some logic so that the check above can be scheduled once every X time, and not just at startup. This way it will work even if the user keeps the Agent running for days. This function could be used in the future for other reasons too, like an auto-update functionality.

There is already an auto-update function in the agent, we may reuse that.

closed 2024-05-08 by MatteoPologruto