arduino/arduino-create-agent #936: Agent does not properly update 'IsOpen' Port State

montymxb, 2024-04-12:
### Describe the problem

The current version of the agent does not appear to update the `IsOpen` state of serial ports. This is set to false by default on adding a new port, but does not appear to be updated when a port is opened, or closed.

This is evident with the latest arduino-create-agent-js-client, which checks this value to determine whether or not to actively close the serial port before attempting to upload a new program via serial. Since this is always false, opening the monitor first effectively prevents the client from uploading anything.

The check is handled inside the method that handles closing the monitor, so invoking it directly also has no effect. The user either has to close all ports forcibly, or directly issue a command to the agent.

### To reproduce

1. Download the latest agent
2. Connect a board (probably flashed with a program that opens up the serial)
3. Open the 'Debug Console' in the drop-down for the agent
4. Run `list` & observe the 'IsOpen' property is false
5. Open the serial port using the appropriate args: ex. `open /dev/MyDeviceXYZ 9600 timed`
6. List again, and observe `IsOpen` is still false

### Expected behavior

As far as I understand (based on the behavior of the JS sdk) opening the serial port to a board should mark the `IsOpen` property as true, and closing it should mark it as false.

I presume such updates could be performed in https://github.com/arduino/arduino-create-agent/blob/main/serialport.go#L270, and https://github.com/arduino/arduino-create-agent/blob/main/serialport.go#L349, using the current `serialPorts`.

### Arduino Create Agent version

1.4

### Operating system

macOS

### Operating system version

14.4

### Browser

Chrome, Firefox

### Browser version

respecitvely 123.0.6312.124, 124.0.2

### Additional context

arm64 setup

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

closed 2024-04-16 by cmaglie

montymxb, 2024-04-24:
Hi all, thanks for resolving this quickly. @Xayton @cmaglie any news on when we can expect a patch release to be made available for this?

Xayton, 2024-04-30:
Hello @montymxb, we plan to integrate an additional fix, related to the recent "board discovery" changes, and then release the 1.4.1 version. This is expected before the end of the current week.