arduino/arduino-create-agent #975: arduino create agent not opening on win7.

arnavvg, 2024-07-23:
### Describe the problem

it is showing me this error.

### To reproduce

i opened the arduino create agent and then it showed me this. it also had the gray arduino symbol.
[image]

### Expected behavior

it should turn on?

### Arduino Create Agent version

ArduinoCreateAgent-1.5.0-windows-386

### Operating system

Windows

### Operating system version

7

### Browser

Chrome (win7)not up to date

### Browser version

109.0.5414.120 (Official Build) (64-bit)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

Xayton, 2024-07-26:
Hello @arnavvg a few things:
- it looks like the agent is crashing on Windows, can you get the crash report and attach them here?
  https://github.com/arduino/arduino-create-agent/wiki/How-to-use-crash-report-functionality
- could you try with the latest version (1.6.0) that was recently released?
- could you try to do a clean installation, that is uninstall and delete the `C:\users\<username>\.arduino-create` folder?
  Do this after getting the crash report, otherwise you will lose the reports.

BallscrewBob, 2024-11-17:
There were IIRC some minor issues with antivirus and antimalware etc. a long time ago that could cause premature closure of the agent. Please make sure you have added it to any "exclusions" lists in that regard. Also "auto roaming" vpns or similar could also cause connection breaks. If you use any such device or software then allow the agent to "pass through" or "DMZ" lists. I would also suggest moving to windows 10 to add more compatibility if that is feasible as win 7 was phased out quite a while ago. I had run the agent under virtual machines but had mixed success as not all VM software works the same way especially when it comes down to both USB devices and networking.