arduino/arduino-create-agent #977: Agent cannot start due to an error while starting or installing "serial-discovery"

Xayton, 2024-07-26:
Environment details: macOS, Agent 1.6.0
Reported by @charlesbones

The agent starts and the closes immediately with a panic, with the following logs:
```
ERRO[0001] Error downloading serial-discovery: invalid character ‘y’ after top-level value
panic: invalid character ‘y’ after top-level value
goroutine 69 [running]:
main.(*SerialPortList).runSerialDiscovery(0x1400027ecc0?)
	/Users/runner/work/arduino-create-agent/arduino-create-agent/serial.go:130 +0x568
main.(*SerialPortList).Run(0x140001a0180?)
	/Users/runner/work/arduino-create-agent/arduino-create-agent/serial.go:118 +0x2c
created by main.loop in goroutine 50
	/Users/runner/work/arduino-create-agent/arduino-create-agent/main.go:410 +0x1244
```

Looking at the message, this might be a `installed.json` corruption problem. See also this previous issue: https://github.com/arduino/arduino-create-agent/pull/635

Here is the content of the JSON file on the same machine (note the wrong data at the end):
```json
{"arduino-fwuploader":"/Users/carlos/.arduino-create/arduino/arduino-fwuploader/2.4.1","arduino-fwuploader-2.4.1":"/Users/carlos/.arduino-create/arduino/arduino-fwuploader/2.4.1","bossac":"/Users/carlos/.arduino-create/arduino/bossac/1.7.0","bossac-1.7.0":"/Users/carlos/.arduino-create/arduino/bossac/1.7.0","bossac-1.9.1-arduino5":"/Users/carlos/.arduino-create/arduino/bossac/1.9.1-arduino5","fwupdater":"/Users/carlos/.arduino-create/arduino/fwupdater/0.1.12","fwupdater-0.1.12":"/Users/carlos/.arduino-create/arduino/fwupdater/0.1.12","rp2040tools":"/Users/carlos/.arduino-create/arduino/rp2040tools/1.0.6","rp2040tools-1.0.6":"/Users/carlos/.arduino-create/arduino/rp2040tools/1.0.6","serial-discovery":"/Users/carlos/.arduino-create/builtin/serial-discovery/1.4.1","serial-discovery-1.4.0":"/Users/carlos/.arduino-create/builtin/serial-discovery/1.4.0","serial-discovery-1.4.1":"/Users/carlos/.arduino-create/builtin/serial-discovery/1.4.1"}y/1.4.1"}
```

umbynos, 2024-08-01:
This affects other OS too, like windows. And causes this error too:
```
get location of {runtime.tools.bossac-1.9.1-arduino5.path}: invalid character '\\' after top-level value
```
A mitigation for now it to reinstall the agent, which causes a deletion of the corrupted json file.

[image]

Xayton, 2024-08-05:
The problem has been fixed by improving the locking code (to avoid file corruption) and by handling the case of an existing installed.json file that is already corrupted.

closed 2024-08-05 by MatteoPologruto