arduino/arduino-create-agent #980: "Esptool: No such file or directory" error when uploading code to ESP8266 board on a Mac

FlynnMarlie, 2024-07-30:
### Describe the problem

Hi, we have begun to notice an issue with the latest release (1.6.0) of the Cloud agent on Macbooks. The issue happens when uploading any code to a ESP8266 board via Arduino Cloud. 

After some investigation, we saw that, when the Mac updates the create agent to version 1.6.0 (the agent was updated automatically on student computers), it leaves the existing application in the applications folder. Students can still go to Arduino Cloud, open a sketch, and select the board. So, initially it looks like the agent is running and working correctly. We can also see that it knows that the board is connected if a student plugs it into their computer. 

However when the student attempts to upload their code to the device we get the following error:

[image]

We navigated to the "esptool" folder in the .arduino-create directory, but found that this folder is empty (it was empty on multiple student computers, all of which ran MacOS). After attempting to place the files it was looking for in the directory manually, it seemed as if the files were getting deleted when the agent attempts to flash the attached device. 

We were able to confirm that the Mac computers we looked at were able to flash to the device and upload the new code by first quitting the agent that was running, then installing the previous version of the agent (version 1.5.0) and then running it. It then allowed us to upload the new code successfully. Going back to version 1.5.0 worked as long as the device didn't restart and the browser wasn't refreshed as a restart or browser refresh caused it to download and use version 1.6.0 again and the issue would happen again. 

Here is a screenshot of how the error was displayed in the console:

[image]

Also, I've attached the bin file mentioned in the error.
[not__working_error.zip](https://github.com/user-attachments/files/16421765/not__working_error.zip)

Any help to solve this issue or finding a better work-around until the next version is released would be greatly appreciated. 

### To reproduce

To reproduce the issue, simply create a new sketch in Arduino Create (https://create.arduino.cc/editor). 

Next, connect an Arduino board to the computer and attempt to upload any code to it. The error should show then. The board we used with students is the ESP8266: https://store-usa.arduino.cc/products/nodemcu-esp8266

Please note that this error only seems to be affecting Mac computers. Uploading code using a Windows computer seems to work fine. 

### Expected behavior

After following the instructions to reproduce the error, the error should show in the Console. 

### Arduino Create Agent version

1.6.0

### Operating system

macOS

### Operating system version

All operating system versions seems to be affected

### Browser

Chrome

### Browser version

All browser versions seems to be affected

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

umbynos, 2024-08-05:
The problem has been introduced in 1.6.0 (probably in #953):

Running the same `curl` request:
``` bash
curl 'http://127.0.0.1:8991/v2/pkgs/tools/installed' -X POST -H 'User-Agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:128.0) Gecko/20100101 Firefox/128.0' -H 'Accept: */*' -H 'Accept-Language: en-US,en;q=0.5' -H 'Accept-Encoding: gzip, deflate, br, zstd' -H 'Content-Type: text/plain;charset=UTF-8' -H 'Origin: https://app.arduino.cc' -H 'Connection: keep-alive' -H 'Sec-Fetch-Dest: empty' -H 'Sec-Fetch-Mode: cors' -H 'Sec-Fetch-Site: cross-site' -H 'Priority: u=4' -H 'Pragma: no-cache' -H 'Cache-Control: no-cache' --data-raw '{"checksum":"SHA-256:bded1dca953377838b6086a9bcd40a1dc5286ba5f69f2372c22a1d1819baad24","name":"esptool","packager":"esp8266","signature":"852b58871419ce5e5633ecfaa72c0f0fa890ceb51164b362b8133bc0e3e003a21cec48935b8cdc078f4031219cbf17fb7edd9d7c9ca8ed85492911c9ca6353c9aa4691eb91fda99563a6bd49aeca0d9981fb05ec76e45c6024f8a6822862ad1e34ddc652fbbf4fa909887a255d4f087398ec386577efcec523c21203be3d10fc9e9b0f990a7536875a77dc2bc5cbffea7734b62238e31719111b718bacccebffc9be689545540e81d23b81caa66214376f58a0d6a45cf7efc5d3af62ab932b371628162fffe403906f41d5534921e5be081c5ac2ecc9db5caec03a105cc44b00ce19a95ad079843501eb8182e0717ce327867380c0e39d2b48698547fc1d0d66","url":"https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-3/x86_64-linux-gnu.esptool-f80ae31.tar.gz","version":"2.5.0-3-20ed2b9"}'
```

with 1.6.0 :arrow_down: 
```
{"name":"fault","id":"8jjT_VHw","message":"Create file /home/umberto/.arduino-create/esp8266/esptool/2.5.0-3-20ed2b9/esptool: open /home/umberto/.arduino-create/esp8266/esptool/2.5.0-3-20ed2b9/esptool: is a directory","temporary":false,"timeout":false,"fault":true}
```

with 1.5.0
```
{"status":"ok"}
```

Xayton, 2024-08-05:
We discovered this is a regression caused by [some changes in 1.6.0](https://github.com/arduino/arduino-create-agent/commit/887b22bbf9804958b3752002e51f20548892766a) and by a peculiar structure of the `esptool` packages. The package has a root folder "esptool" with the same name of the file contained in it ("esptool").

closed 2024-08-05 by umbynos