arduino/arduino-create-agent #1000: `open` command fails to connect to boards with CH340 USB chip on Windows

per1234, 2024-12-12:
### Describe the problem

The WCH CH340 USB to serial bridge chip is common on boards used by the Arduino community. The manufacturer of this chip recently released a new version of the Windows driver, which users are receiving via a Windows update.

🐛 The port is spuriously closed immediately after an `open` command under the following conditions:

- The port is produced by a CH340 USB chip.
- The user is using the Windows operating system
- The user has the latest CH340 driver version (3.9.2024.9) installed.

### To reproduce

#### Equipment

- Any device that has a WCH CH340 USB chip.
- A Windows machine.

#### Steps

1. Connect the device to your computer with a USB cable.
1. Open Windows **Device Manager**.
1. Find the device under the "**Ports (COM & LPT)**" section of the **Device Manager** tree.
1. Right click on the device.
   A context menu will open.
1. Select "**Properties**" from the menu.
   The "**USB-SERIAL CH340 (COM_n_) Properties**" dialog will open.
1. Select the "**Driver**" tab in the dialog.
1. Verify that version `3.9.2024.9` is shown in the "**Driver Version**" field of the dialog.
1. Close the dialog and the **Device Manager** window.
1. Start **Arduino Cloud Agent**.
   An icon will appear in the Windows "[notification area](https://support.microsoft.com/en-us/windows/customize-the-taskbar-notification-area-e159e8d2-9ac5-b2bd-61c5-bb63c1d437c3)".
1. Click the **Arduino Cloud Agent** icon in the notification area.
   A menu will open.
1. Select "**Open Debug Console**" from the menu.
   The "**Arduino Cloud Agent Debug Console**" page will open in your browser.
1. Type `open <port name> 9600` in the field at the bottom of the page.
   (where `<port name>` is the name of the serial port produced by the CH340 device)

🐛 The port is opened, but then immediately closed:

```text
open COM17 9600

{
  "Cmd": "Open",
  "Desc": "Got register/open on port.",
  "Port": "COM17",
  "Baud": 9600,
  "BufferType": "default"
}

{
  "Cmd": "Close",
  "Desc": "Got unregister/close on port.",
  "Port": "COM17",
  "Baud": 9600
}

Shutting down writer on COM17

writerBuffered just got closed. make sure you make a new one. port:COM17
```

### Expected behavior

Port remains open until disconnected or a `close` command is executed.

### Cloud Agent version

d36d0e142b8b46c0cde117f94b37b37422b806d7

### Operating system

- Windows

### Operating system version

- Windows 11

### Additional context

I found that the disconnection is caused by this code:

https://github.com/arduino/arduino-create-agent/blob/d36d0e142b8b46c0cde117f94b37b37422b806d7/serialport.go#L158-L165

Here is a derived minimal Go program that can be used to reproduce the fault:

```go
package main

import (
	"log"
	"time"

	"go.bug.st/serial"
)

const portName string = "COM17"

func main() {
	mode := &serial.Mode{
		BaudRate: 9600,
	}
	port, err := serial.Open(portName, mode)
	if err != nil {
		log.Fatal(err)
	}

	timeCheckOpen := time.Now()
	buff := make([]byte, 100)
	for {
		n, err := port.Read(buff)
		if err != nil {
			log.Fatal(err)
			break
		}
		if n <= 0 {
			diff := time.Since(timeCheckOpen)
			if diff.Nanoseconds() < 1000000 {
				log.Println("Too soon since last empty read")
				break
			}
			timeCheckOpen = time.Now()
		}
	}
}
```

---

This problem causes the [Arduino Cloud Editor **Serial Monitor**](https://docs.arduino.cc/arduino-cloud/guides/editor/#serial-monitor) to fail to connect to the port, with the following error notification:

> **We can’t find the device, please make sure it is connected via serial port.
You can still see log history here, but data won’t be printed until new connection.**

---

The fault does not occur after I install the older versions 3.8.2023.2 and 3.7.2022.1 of the CH340 driver.

---

The fault does not occur when using [`arduino-cli monitor`](https://arduino.github.io/arduino-cli/dev/commands/arduino-cli_monitor/).

---

The fault does not occur on my macOS and Linux machines.

---

The fault does not occur when using boards with other USB chips (FTDI FT232R, Silicon Labs CP2104, ATmega16U2 w/ [Arduino firmware](https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/atmegaxxu2)).

---

Originally reported at https://forum.arduino.cc/t/error-when-opening-serial-monitor-we-cant-find-the-device/1331419/1

#### Additional reports:

- https://forum.arduino.cc/t/error-when-opening-serial-monitor-we-cant-find-the-device/1331419/3

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

Reginald-Gillespie, 2024-12-15:
I was investigating the same issue - it makes sense that it was a CH340 driver update that broke it, was confused for a bit since the ACA hadn't had any updates pushed recently.

per1234, 2024-12-16:
> this impacts the classic Arduino Nano

I am not able to reproduce the fault using a [classic **Arduino Nano**](https://docs.arduino.cc/hardware/nano). From my tests, the fault is specific to the boards that have a CH340 USB chip. The official **Arduino Nano** uses the FTDI FT232 USB chip, not the CH340. However, the cheap Nano derivative boards do use the CH340 and I confirm that the fault affects those boards.

Reginald-Gillespie, 2024-12-17:
Oh yes, I was looking at my off-brand boards. I assumed they were high enough quality to be exact clones, but I didn't check, mb.

cmaglie, 2025-03-17:
Should be fixed by: https://github.com/bugst/go-serial/pull/200

cmaglie, 2025-03-18:
There are test builds of the agent here (at the bottom of the page): https://github.com/arduino/arduino-create-agent/actions/runs/13904990365
@Reginald-Gillespie (or anyone else) may I ask you to try it?

Reginald-Gillespie, 2025-03-18:
@cmaglie built it yesterday, can confirm it works for me :+1:

PR #1023 (2025-03-19): Updated go-serial library to v1.6.3
**Please check if the PR fulfills these requirements**

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-create-agent/pulls)
      before creating one)
- [ ] Tests for the changes have been added (for bug fixes / features)

**What kind of change does this PR introduce?**
Updated go-serial library to fix #1000.

**What is the current behavior?**
Agent is not able to open CH340-based serial ports

**What is the new behavior?**
Agent should be able to open CH340-based serial ports

**Does this PR introdu [... cut]
closed 2025-03-19 by cmaglie