arduino/arduino-create-agent #1017: Installer ignoring --prefix to specify install directory

elopez-peach, 2025-02-14:
### Describe the problem

Running `ArduinoCloudAgent.exe --help` opens a dialogue box that gives details on install arguments. 
This window states that the `--prefix` argument can be used to specify an alternate install directory. It also states that the default install directory is `C:\Program Files (x86)/ArduinoCloudAgent`.
Instead, the installer always installs to the `%appdata%\ArduinoCloudAgent\` folder for the user that is running the installer.

We are a school trying to use PDQ Deploy to install the cloud agent to all of our lab computers silently.

### To reproduce

1. Run installer with argument `--help` to get the list of arguments
2. Attempt to perform a silent install using the arguments provided `--mode unattended --prefix "C:/Program Files/ArduinoCloudAgent"`
3. Install directory that was specified does not exist. Installer used `%appdata%` for the user running the installer instead

### Expected behavior

Expected installer to use the directory specified in the installer command so that it would be available to all users

### Arduino Create Agent version

1.7.0

### Operating system

Windows

### Operating system version

10/11

### Browser

Chrome

### Browser version

133

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-create-agent/issues?q=)
- [x] I verified the problem still occurs when using the latest version
- [x] My report contains all necessary details

tscott-hoy, 2025-05-02:
+1 - also still prevalent as of 1.7.1-rc1.

derdegenhard, 2025-05-04:
+1
Please let us install it unattended in a specific folder on Windows.

rodrigolangella, 2026-02-12:
+1  -  Need to deploy via Intune to multiple devices and I can't find a solution for this